Koenkk/zigbee2mqtt #17519: TS110E_1gang_1 device turns on with brightness 0

emimor123, 2023-05-01:
### What happened?

When the TS110E_1gang_1 device is turned off with the physical switch, the brightness value drops to 0 but when it is turned on it does not maintain the previous value, it remains at 0, so even though the device is on, the light remains off brightness=0
[node-red debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/11364899/node-red.debug.txt)

### What did you expect to happen?

When the device is turned off brightness = 0 , when the device is turned on brightness = "last state" regardless of where it is turned on or off.

### How to reproduce it (minimal and precise)

Step 1: Power off the device from the physical switch.
Step 2: Regardless of where you try to turn the light on (switch, alexa, or HA) the device turns on with brightness = 0 bulb off.
![Imagen de WhatsApp 2023-05-01 a las 12 21 07](https://user-images.githubusercontent.com
[image]
/86189076/235440223-e8ac9d0a-43da-4434-8615-1e2122d3672c.jpg)

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

 20221226

### Adapter

zStack3x0 Cod.m Lan coordinator

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/11364957/log.txt)

emimor123, 2023-06-03:
Hi, nobody? Problem still continues 
Regardless of where you try to turn the light on (switch, alexa, or HA) the device turns on with brightness = 0 bulb off.
Them is necessary to change brightness with app or telling alexa. the previous state is never saved.

closed 2023-07-12 by github-actions[bot]

jonathh21, 2023-08-14:
I have exactly this issue.

Did the OP resolve it??