Koenkk/zigbee2mqtt #17521: [New device support]: _TZE204_dzuqwsyg

Sherk737, 2023-05-01:
### Link

https://www.aliexpress.com/item/1005004722220463.html?spm=a2g0o.order_list.order_list_main.5.18751802e7UEar

### Database entry

{     "last_seen": "2023-04-30T15:58:28+04:00",     "linkquality": 76 }

### Comments

I bought couple of thermostat, some got recongnize and some not. the difrent that some got Zigbee Model "TS0601" and Zigbee Manufacturer " _TZE200_dzuqwsyg" this one got supported and recongnize by z2m the other one that got not supported is Zigbee Model "TS0601" and Zigbee Manufacturer "_TZE204_dzuqwsyg". I tried to add it manually by copying a convertor I found it to the supported one and I change the zigbee manuacturer but it added but not working propally. please add support for it.

Home Assistant : 2023.4.6
Zigbee2MQTT : 1.30.3-1

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const ota = require('zigbee-herdsman-converters/lib/ota');
const {repInterval, defaultBindGroup, OneJanuary2000} = require('zigbee-herdsman-converters/lib/constants');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fz_moes_thermostat_custom = {
	cluster: 'manuSpecificTuya',
	type: ['commandGetData', 'commandSetDataResponse', 'commandDataResponse'],
	convert: (model, msg, publish, options, meta) => {
            const dpValue = tuya.firstDpValue(msg, meta, 'moes_thermostat');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);
		let temperature;
		/* See tuyaThermostat above for message structure comment */
		switch (dp) {
		case tuya.dataPoints.moesSchedule:
			return {
				program: [
					{weekdays_p1: value[0] + 'h:' + value[1] + 'm ' + value[2]/2 + '°C'},
					{weekdays_p2: value[3] + 'h:' + value[4] + 'm ' + value[5]/2 + '°C'},
					{weekdays_p3: value[6] + 'h:' + value[7] + 'm ' + value[8]/2 + '°C'},
					{weekdays_p4: value[9] + 'h:' + value[10] + 'm ' + value[11]/2 + '°C'},
					{saturday_p1: value[12] + 'h:' + value[13] + 'm ' + value[14]/2+ '°C'},
					{saturday_p2: value[15] + 'h:' + value[16] + 'm ' + value[17]/2 + '°C'},
					{saturday_p3: value[18] + 'h:' + value[19] + 'm ' + value[20]/2 + '°C'},
					{saturday_p4: value[21] + 'h:' + value[22] + 'm ' + value[23]/2 + '°C'},
					{sunday_p1: value[24] + 'h:' + value[25] + 'm ' + value[26]/2 + '°C'},
					{sunday_p2: value[27] + 'h:' + value[28] + 'm ' + value[29]/2 + '°C'},
					{sunday_p3: value[30] + 'h:' + value[31] + 'm ' + value[32]/2 + '°C'},
					{sunday_p4: value[33] + 'h:' + value[34] + 'm ' + value[35]/2 + '°C'},
				],
			};
		case 1: // system mode (OFF, ON)
			if(value == 0) {
				return {state: 'OFF'};
			} else if(value == 1) {
				return {state: 'ON'};
			} else {
				return {};
			}
		case 2: // system mode (cooling, heating, fan)
			if(value == 0) {
				return {system_mode: 'cool'};
			} else if(value == 1) {
				return {system_mode: 'heat'};
			} else if(value == 2) {
				return {system_mode: 'fan_only'};
			} else {
				return {};
			}
		case 28: // fan mode
			if(value == 0) {
				return {fan_mode: 'low'};
			} else if(value == 1) {
				return {fan_mode: 'medium'};
			} else if(value == 2) {
				return {fan_mode: 'high'};
			} else if(value == 3) {
				return {fan_mode: 'auto'};
			} else {
				return {};
			}
		case 16: // temperature setpoint for both heating and cooling
			return {occupied_heating_setpoint: value};
		case tuya.dataPoints.moesChildLock:
			return {child_lock: value ? 'LOCK' : 'UNLOCK'};
		case tuya.dataPoints.moesMaxTempLimit:
			return {max_temperature_limit: value};
		case tuya.dataPoints.moesMaxTemp:
			return {max_temperature: value};
		case tuya.dataPoints.moesDeadZoneTemp:
			return {deadzone_temperature: value};
		case tuya.dataPoints.moesLocalTemp:
			temperature = value & 1<<15 ? value - (1<<16) + 1 : value;
			return {local_temperature: parseFloat((temperature / 10).toFixed(1))};
		case tuya.dataPoints.moesTempCalibration:
			temperature = value;
			// for negative values produce complimentary hex (equivalent to negative values)
			if (temperature > 4000) temperature = temperature - 4096;
			return {local_temperature_calibration: temperature};
		case tuya.dataPoints.moesHold: // state is inverted, preset_mode is deprecated
			return {preset_mode: value ? 'program' : 'hold', preset: value ? 'program' : 'hold'};
		case tuya.dataPoints.moesScheduleEnable: // state is inverted, preset_mode is deprecated
			return {preset_mode: value ? 'hold' : 'program', preset: value ? 'hold' : 'program'};
		case tuya.dataPoints.moesValve:
			return {heat: value ? 'OFF' : 'ON'};
		case tuya.dataPoints.moesSensor:
			switch (value) {
			case 0:
				return {sensor: 'IN'};
			case 1:
				return {sensor: 'AL'};
			case 2:
				return {sensor: 'OU'};
			default:
				return {sensor: 'not_supported'};
			}
		default: // DataPoint 17 is unknown
			meta.logger.warn(`zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP #${
				dp} with data ${JSON.stringify(msg.data)}`);
		}
	},
};

const tz_state = {
	key: ['state'],
	convertSet: async (entity, key, value, meta) => {
		if(value == "OFF") {
			await tuya.sendDataPointBool(entity, 1, false);
		} else if(value == "ON") {
			await tuya.sendDataPointBool(entity, 1, true);
		}
	},
};

const tz_system_mode = {
	key: ['system_mode'],
	convertSet: async (entity, key, value, meta) => {
		if(value == "cool") {
			await tuya.sendDataPointEnum(entity, 2, 0);
		} else if(value == "heat") {
			await tuya.sendDataPointEnum(entity, 2, 1);
		} else if(value == "fan_only") {
			await tuya.sendDataPointEnum(entity, 2, 2);
		}
	},
};

const tz_temperature_setpoint = {
	key: ['occupied_heating_setpoint'],
	convertSet: async (entity, key, value, meta) => {
		await tuya.sendDataPointValue(entity, 16, value);
	},
};

const tz_fan_mode = {
	key: ['fan_mode'],
	convertSet: async (entity, key, value, meta) => {
		if(value == "low") {
			await tuya.sendDataPointEnum(entity, 28, 0);
		} else if(value == "medium") {
			await tuya.sendDataPointEnum(entity, 28, 1);
		} else if(value == "high") {
			await tuya.sendDataPointEnum(entity, 28, 2);			
		} else if(value == "auto") {
			await tuya.sendDataPointEnum(entity, 28, 3);			
		}
	},
};

module.exports = [
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE204_dzuqwsyg'}],
        model: 'BAC-002-ALZB',
        vendor: 'HKGK',
        description: 'BAC series thermostat',
        fromZigbee: [fz_moes_thermostat_custom],
        toZigbee: [tz_state, tz_system_mode, tz_fan_mode, tz_temperature_setpoint, tz.moes_thermostat_child_lock, tz.moes_thermostat_current_heating_setpoint, tz.moes_thermostat_mode,
            tz.hgkg_thermostat_standby, tz.moes_thermostat_sensor, tz.moes_thermostat_calibration,
            tz.moes_thermostat_deadzone_temperature, tz.moes_thermostat_max_temperature_limit],
        exposes: [e.child_lock(), e.deadzone_temperature(), e.max_temperature_limit(),
			exposes.switch()
				.withState('state', false, 'OFF/ON state'),
            exposes.climate()
				.withSetpoint('occupied_heating_setpoint', 16, 35, 1, ea.STATE_SET)
                .withLocalTemperature(ea.STATE)
				.withLocalTemperatureCalibration(ea.STATE_SET)
                .withSystemMode(['cool', 'heat', 'fan_only'], ea.STATE_SET)
				.withFanMode(['low', 'medium', 'high', 'auto'], ea.STATE_SET)
				.withRunningState(['idle', 'heat', 'cool'], ea.STATE)
                .withPreset(['hold', 'program'])],
        onEvent: tuya.onEventSetLocalTime,
    },
];
```

Sherk737, 2023-05-12:
external convertal not working could you add it to Z2M

ant0nkr, 2023-06-13:
Is there any plan to add support of _TZE204_dzuqwsyg thermostat?
Converter stopped working with version 1.31.2-1.

closed 2023-07-23 by github-actions[bot]

piposant, 2023-09-14:
I am very interested in this thermostat but I can't get it to work.

pippocuce, 2023-11-18:
Please add support for this thermostat thanks

aabrodskiy, 2023-12-12:
I've bought a few of these thermostats and ran into the same issue. Interestingly, some of the devices got recognised as a different supported manufacturer, but most of them didn't. 
Has anyone found a way to make them work?

saltandvinegarcrisps, 2024-01-12:
I managed to get most controls working with

```js
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const extend = require('zigbee-herdsman-converters/lib/extend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const ota = require('zigbee-herdsman-converters/lib/ota');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const utils = require('zigbee-herdsman-converters/lib/utils');
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

// Device: https://www.aliexpress.com/item/1005004174932182.html
// Very similar datapoints: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts#L6920
const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_dzuqwsyg',
        },
    ],
    model: 'TYBAC-001',
    vendor: 'TuYa',
    description: 'Central Air Conditioner Thermostat Temperature Controller',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        e.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('Turn the thermostat ON/OFF'),
        e.climate()
            .withLocalTemperature(ea.STATE)
            .withSetpoint('current_heating_setpoint', 5, 35, 1, ea.STATE_SET)
            .withSystemMode(['cool', 'heat', 'fan_only'], ea.STATE_SET)
            .withFanMode(['low', 'medium', 'high', 'auto'], ea.STATE_SET)
        e.binary('manual_mode', ea.STATE_SET, 'ON', 'OFF').withDescription('Manual = ON or Schedule = OFF'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.onOff],
            [2, 'system_mode', tuya.valueConverterBasic.lookup({'cool': tuya.enum(0), 'heat': tuya.enum(1), 'fan_only': tuya.enum(2)})],
            [16, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
            [24, 'local_temperature', tuya.valueConverter.divideBy10],
            [28, 'fan_mode', tuya.valueConverterBasic.lookup({'low': tuya.enum(0), 'medium': tuya.enum(1), 'high': tuya.enum(2), 'auto': tuya.enum(3)})],
            [101, 'manual_mode', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;
```

pippocuce, 2024-01-13:
> Sono riuscito a far funzionare la maggior parte dei controlli con
> 
> ```js
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const globalStore = require('zigbee-herdsman-converters/lib/store');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const utils = require('zigbee-herdsman-converters/lib/utils');
> const {} = require('zigbee-herdsman-converters/lib/modernExtend');
> const {} = require('zigbee-herdsman-converters/lib/tuya');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> // Device: https://www.aliexpress.com/item/1005004174932182.html
> // Very similar datapoints: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts#L6920
> const definition = {
>     fingerprint: [
>         {
>             modelID: 'TS0601',
>             manufacturerName: '_TZE204_dzuqwsyg',
>         },
>     ],
>     model: 'TYBAC-001',
>     vendor: 'TuYa',
>     description: 'Central Air Conditioner Thermostat Temperature Controller',
>     fromZigbee: [tuya.fz.datapoints],
>     toZigbee: [tuya.tz.datapoints],
>     onEvent: tuya.onEventSetTime,
>     configure: tuya.configureMagicPacket,
>     exposes: [
>         e.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('Turn the thermostat ON/OFF'),
>         e.climate()
>             .withLocalTemperature(ea.STATE)
>             .withSetpoint('current_heating_setpoint', 5, 35, 1, ea.STATE_SET)
>             .withSystemMode(['cool', 'heat', 'fan_only'], ea.STATE_SET)
>             .withFanMode(['low', 'medium', 'high', 'auto'], ea.STATE_SET)
>         e.binary('manual_mode', ea.STATE_SET, 'ON', 'OFF').withDescription('Manual = ON or Schedule = OFF'),
>     ],
>     meta: {
>         tuyaDatapoints: [
>             [1, 'state', tuya.valueConverter.onOff],
>             [2, 'system_mode', tuya.valueConverterBasic.lookup({'cool': tuya.enum(0), 'heat': tuya.enum(1), 'fan_only': tuya.enum(2)})],
>             [16, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
>             [24, 'local_temperature', tuya.valueConverter.divideBy10],
>             [28, 'fan_mode', tuya.valueConverterBasic.lookup({'low': tuya.enum(0), 'medium': tuya.enum(1), 'high': tuya.enum(2), 'auto': tuya.enum(3)})],
>             [101, 'manual_mode', tuya.valueConverter.onOff],
>         ],
>     },
> };
> 
> module.exports = definition;
> ```

Hi, very remarkable, I also have this thermostat, how do I enter this code and where?
Can you help me?
I'm Italian and I didn't understand the official instructions well, can you explain the steps I need to do?
Thank you

saltandvinegarcrisps, 2024-01-13:
@pippocuce Follow this guide https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

pippocuce, 2024-01-14:
> Segui questa guida https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

ho seguito ma non so come salvare la definizione esterna generata

pippocuce, 2024-02-01:
It works for me with this configuration:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE204_dzuqwsyg'}],
    model: 'BAC-003',
    vendor: 'TuYa',
    description: 'Central Air Conditioner Thermostat Temperature Controller',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        e.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('Turn the thermostat ON/OFF'),
        e.climate()
                .withSetpoint('current_heating_setpoint', 16, 30, 1, ea.STATE_SET)
                .withLocalTemperature(ea.STATE)
                .withSystemMode(['off', 'cool', 'heat', 'fan_only'], ea.STATE_SET)
                .withFanMode(['low', 'medium', 'high', 'auto'], ea.STATE_SET),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.onOff],
            [2, 'system_mode', tuya.valueConverterBasic.lookup({'cool': tuya.enum(0), 'heat': tuya.enum(1), 'fan_only': tuya.enum(2)})],
            [16, 'current_heating_setpoint', tuya.valueConverterBasic.divideBy(1)],
            [24, 'local_temperature', tuya.valueConverter.divideBy10],
            [28, 'fan_mode', tuya.valueConverterBasic.lookup({'low': tuya.enum(0), 'medium': tuya.enum(1), 'high': tuya.enum(2), 'auto': tuya.enum(3)})],
        ],
    },
};

module.exports = definition;
```

Koenkk, 2024-02-02:
@pippocuce would you mind creating a [PR](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts) for this?

pippocuce, 2024-02-02:
I'm sorry but I'm Italian, what does PR mean?

Koenkk, 2024-02-03:
@pippocuce pull request, I've added it directly now, thanks!

Integrated the changes, you can remove the external converter after updating to the next release.

pippocuce, 2024-02-04:
Thank you for considering my effort

saltandvinegarcrisps, 2024-07-03:
> @pippocuce pull request, I've added it directly now, thanks!
> 
> Integrated the changes, you can remove the external converter after updating to the next release.

Thank You! :pray: