Koenkk/zigbee2mqtt #17524: [Wrong device]: Tuya Smoke detector 2 wrong image

henkiejan1, 2023-05-01:
### Link

https://aliexpress.com/item/1005003970944017.html

### Model

PA-44Z

### Description

Smoke Dector

### Vendor

Tuya

### Picture (link)

https://i5.walmartimages.com/asr/eb545b22-4149-47f7-972b-9e53642f7840.c71d5eff2e1bc3dbfa2328f6a04637da.jpeg

### Database entry

{"id":43,"type":"EndDevice","ieeeAddr":"0xa4c1382d31e90d3a","nwkAddr":53937,"manufId":4417,"manufName":"_TZE200_m9skfctm","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�~�+\u0011","65506":56,"65508":0,"65534":0,"appVersion":72,"stackVersion":0,"dateCode":"","manufacturerName":"_TZE200_m9skfctm","zclVersion":3,"modelId":"TS0601","powerSource":3}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1682964363461,"defaultSendRequestWhen":"immediate"}

### Notes

Looks like white label but i if search on Alibaba i found this smoke detector with the model PA-44Z (whereby i think Z means Zigbee version). But mostly the picture is not relevant standard.

Koenkk, 2023-05-02:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Koenkk, 2023-10-06:
@jvcsw added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

jvcsw, 2023-10-07:
Thank you so much