Koenkk/zigbee2mqtt #17525: [Wrong device]: Tuya Human Presence Detector wrong Picture / Model

KrX3D, 2023-05-01:
### Link

https://de.aliexpress.com/item/1005004836104902.html?spm=a2g0o.productlist.main.3.263b5606YzomoN&algo_pvid=c56c52b2-48b4-4a16-8818-d0662123bb13&algo_exp_id=c56c52b2-48b4-4a16-8818-d0662123bb13-1&pdp_npi=3%40dis%21EUR%2140.96%2122.53%21%21%21%21%21%402100b18f16829663973277073d078a%2112000030676950026%21sea%21DE%211798357663&curPageLogUid=7jEHGXt91h3z

### Model

ZY-M100-S

### Description

Human Presence Detector

### Vendor

Tuya

### Picture (link)

https://ae01.alicdn.com/kf/S28430e89d4284bfc8e7afd342caefab87.jpg

### Database entry

{"id":9,"type":"Router","ieeeAddr":"0xdc8e95fffe2bb4aa","nwkAddr":65176,"manufId":4098,"manufName":"_TZE204_ztc6ggyl","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"k��+i","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_ztc6ggyl","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1682966088284,"defaultSendRequestWhen":"immediate"}

### Notes

The Picture shows the ceiling model:

Model ZY-M100-L

my device is ZY-M100-S

Koenkk, 2023-05-02:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-05-02 by Koenkk

icuMedia, 2023-05-03:
Is it worth waiting for it to be on the main branch?

icuMedia, 2023-05-03:
I have used the dev branch and it pairs, but then is recognized as: 
`"TuYa","model":"Curtain motor/roller blind motor/window pusher/tubular motor (TS0601_cover_1)".`

And then i just gives this type of error in the logs:
`TuYa_cover_control: Unhandled DP #109 for _TZE204_sxm7l9xa: {"dp":109,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,71]}}`

Koenkk, 2023-05-04:
@Martijndegeus please post the data/database.db entry of your device. Note that the picture will only be fixed after the next release (description/vendor/model should be correct though)

Martijndegeus, 2023-05-04:
Where can I find the database.db file in home assistant?

Koenkk, 2023-05-04:
- to fix that I need the data/database.db entry
- as what device should it be recognised, https://www.zigbee2mqtt.io/devices/TS0601_smart_human_presence_sensor.html?

Martijndegeus, 2023-05-04:
That's correct, I just paired it again but I don't see it in the database.db.

Koenkk, 2023-05-04:
@richardwonka you added `_TZE204_sxm7l9xa` to TS0601_cover_1 in  https://github.com/Koenkk/zigbee-herdsman-converters/pull/5714, is that correct? @Martijndegeus claims it should be a TS0601_human_presence_sensor

Martijndegeus, 2023-05-04:
This seems to be related: https://github.com/Koenkk/zigbee2mqtt/issues/17502

richardwonka, 2023-05-04:
My bad. Yes, it should be a human presence sensor. Richard —Throw off the bow lines! Dream! Discover!

Koenkk, 2023-05-05:
Let's continue in #17502