Koenkk/zigbee2mqtt #17527: Ver 1.30.4 - TS0505B disaster

tomiisp, 2023-05-01:
### What happened?

After update to 1.30.4 Lidl light TS0505B based -  _TZ3210_c0s1xloa goes completely crazy:
- Color X/Y changed to Color HS
- options effect, do_not_disturb, color_power_on_behavior disappear at all
- refresh of brightness produce error - Publish 'get' 'brightness' to XXX failed: 'Error: Cluster 'lightingColorCtrl' has no attribute 'currentLevel'' 
- refresh of color_temp - Publish 'get' 'color_temp' to XXX failed: 'Error: Cluster 'lightingColorCtrl' has no attribute 'currentLevel''
- refresh of color_hs - Publish 'get' 'color' to XXX failed: 'Error: Cluster 'lightingColorCtrl' has no attribute 'currentLevel''
Works only ON/OFF, setting brightness or light color/temp from HA completely non functional, in previous version this light works fine although it is TS0505B - notorious Z2M problem with Lidl lights based on TS0505B... 

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

N/A

### Adapter

CC2562

Koenkk, 2023-05-02:
What was the previous known working verion?

tomiisp, 2023-05-02:
@Koenkk I running two instances (one in docker) but as I remember it was version 1.30.3

tomiisp, 2023-05-02:
@Koenkk 
UPDATE: seems moving brightness slider, color temp slider and clicking on H/S color works in Z2M web, it control light parameters BUT refresh button produce "has no attribute" error and existing HA automation only turn light On/Off, transition and brightness values set in automation are ignored by Z2M.

Koenkk, 2023-05-03:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0505B', manufacturerName: '_TZ3210_p9ao60da'}],
    model: 'HG08008',
    vendor: 'Lidl',
    description: 'Livarno Home LED ceiling light',
    extend: tuya.extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500], noConfigure: true}),
    configure: async (device, coordinatorEndpoint, logger) => {
        device.getEndpoint(1).saveClusterAttributeKeyValue('lightingColorCtrl', {colorCapabilities: 29});
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

tomiisp, 2023-05-03:
@Koenkk  seems here we mix/swap two issues together, this issue was related about light  _TZ3210_c0s1xloa (model 399629_2110) and not _TZ3210_p9ao60da (model HG08008) - this light stay untouched in latest update but still have that another issue with "flashing" and your converted is for _TZ3210_p9ao60da.
If I apply your converter as is then it change nothing,
If I change in your converter code:
- fingerprint from "_TZ3210_p9ao60da" to "_TZ3210_c0s1xloa"
- model from "HG08008" to "399629_2110"
- description to :  "Livarno Lux Ceiling Panel RGB+CCT"
 then it fix this issue (functionality return back as it was in 1.30.3) and it back works fine with HA

Koenkk, 2023-05-04:
Pushed the fix, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-05-04 by Koenkk