Koenkk/zigbee2mqtt #17530: HA Linkquality Entity Auto-Enabled

jeremywillans, 2023-05-02:
### What happened?

Every time I add a new device to Z2M the entity for linkquality is disabled by default in Home Assistant

Looking in MQTT, this has the attribute `"enabled_by_default":false` - can this be changed?

### What did you expect to happen?

LQ entity is enabled when a new device is added

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

0x26580700

### Adapter

ConBee2

closed 2023-06-09 by github-actions[bot]

jeremywillans, 2023-06-09:
Found the solution for anyone else wanting to implement this.
```
device_options:
  homeassistant:
    linkquality:
      enabled_by_default: true

MacDada, 2024-04-14:
```yaml
# does not work
device_options:
  homeassistant:
    linkquality:
      enabled_by_default: true
```

Actually, it doesn't work for me. MQTT Explorer is showing me that for `linkquality` (and `voltage`, for example) Z2M is still sending `enabled_by_default: false`.

Only after I set it "manually" on each single device, it starts to send `enabled_by_default: true` -> making the entity enabled in HA (as the "initial" value: nothing changes unless you change an entity id as well).

```yaml
# works
devices:
  '0xa4c138413cec6e24':
    friendly_name: BlahBlah
    homeassistant:
      linkquality:
        enabled_by_default: true
```

Zigbee2MQTT version
[1.35.3](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.3) commit: [fe0742a](https://github.com/Koenkk/zigbee2mqtt/commit/fe0742a)
Coordinator type
zStack3x0
Coordinator revision
20221226
Frontend version
0.6.156
zigbee_herdsman_converters_version
18.28.3
zigbee_herdsman_version
0.33.8