Koenkk/zigbee2mqtt #17532: CRASH on interview

corporategoth, 2023-05-02:
### What happened?

I'm trying to interview a set of blinds that had become unresponsive (removed and re-added to the network).  Every time I interview it, I get the following crash:

Error: Failure send findKeyTableEntry:{"type":"Buffer","data":[154,0,1,117,0,156,123,159,254,255,249,227,180,1]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:551:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

### What did you expect to happen?

The interview to succeed, without crashing zigbee2mqtt!

### How to reproduce it (minimal and precise)

Remove device.
Re-add device

CRASH!

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

6.10.3.0 build 297

### Adapter

ezsp (Sonoff Dongle-E)

### Debug log

Error: Failure send findKeyTableEntry:{"type":"Buffer","data":[154,0,1,117,0,156,123,159,254,255,249,227,180,1]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:551:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

franssonjohan, 2023-05-03:
Having the same issue. Updated to Zigbee2MQTT version 1.30.4-1, but the issue is still there.

kirovilya, 2023-05-03:
Can you record herdsman-log for crash moment? https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee-herdsman-debug-logging

corporategoth, 2023-05-03:
I tried once, and then it didn't crash once I had debugging on (also the debug logging can cause other issues for me, as I have a large network, so a lot of traffic, and debugging can slow things down enough that I get timeouts).

When I try and add my next device (likely tonight) I'll see if it crashes again and try again to debug it with herdsman.

franssonjohan, 2023-05-07:
I enable herdsman-logs and two of three devices (Sonoff SNZB-01) re-joined correctly without crashing zigbee2mqtt. Third one did crash it again. I didn't manage to see any error from log output in addons in HomeAssistant before it crashed and log output was refreshed. What's the easiest way to capture these logs running zigbee2mqtt addon under HASS.IO?

franssonjohan, 2023-05-08:
I was just trying to add a new Neo wall plug, with Herdsman logs enabled. Crashed right away, but log output window in the addon section are getting cleared quite fast before I could capture them manually. 

Maybe I'm missing something here but checking the docs for Zigbee2mqtt (https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee-herdsman-debug-logging) it clearly state the  command "docker logs CONTAINER_ID > log.txt 2>&1" for capturing the logs under HomeAssistantOS. But when running under Home Assistant OS with the SSH addon you cannot actually get "outside" the HASSOS docker, right? docker command doesn't exists within a container?

corporategoth, 2023-05-11:
When I DON'T have herdsman logs set to debug, I can reproduce this issue trying to add any new zigbee device.

When I enable debug herdsman logging, the error goes away and pairing is successful.  @Koenkk Any better way to debug this issue?  It's debilitating, and having to enable herdsman debug logging to pair something is not really a solution (plus, I don't like problems that go away when debugging is turned on).

Koenkk, 2023-05-12:
@corporategoth try if it also occurs with different levels of herdsman debug logging, now you are probably using `DEBUG=zigbee-herdsman:*` but try with e.g. `DEBUG=zigbee-herdsman:adapter:ezsp:debg`

Kirmas, 2023-06-12:
Have similar crash but on tuya-T6E. Zigbee2MQTT instaled [using this instruction](https://blakadder.com/t6e-z2m/).
Interesting notes: 

1) I be able to connect new device to the network but after 5-6 tries. 
2) If I reset network and start from the begining first 10 -15 device conecting without any problem. 

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/11728308/log.txt) If I can help somehow to fix this, tell please.

closed 2023-07-23 by github-actions[bot]

lruppert, 2023-08-09:
This is still present in [1.32.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.32.2) commit: [1ec1e57](https://github.com/Koenkk/zigbee2mqtt/commit/1ec1e57)
zigbee2mqtt docker container crashes with the error message in the description, same ezsp.ts and queue.ts line numbers.

oversc0re, 2023-08-13:
I am also on version 1.32.2 (commit #1ec1e57). I can pair fist few devices without a problem. After approx 5 paired devices I start getting this error. When retrying the same device over and over again, sometimes the pairing succeeds, but it can take 10 or more attempts.

(Error: Failure send findKeyTableEntry:{"type":"Buffer","data":[60,0,1,117,0,227,177,224,1,0,141,21,0,1]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:562:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

corporategoth, 2023-09-02:
```
2023-09-02T01:37:25.243Z zigbee-herdsman:adapter:ezsp:debg processMessage: {"messageType":0,"apsFrame":{"profileId":260,"clusterId":2820,"sourceEndpoint":1,"destinationEndpoint":1,"options":64,"groupId":0,"sequence":56},"lqi":156,"rssi":-61,"sender":41270,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[24,14,10,11,5,41,242,2]}}
2023-09-02T01:37:26.965Z zigbee-herdsman:adapter:ezsp:debg Device join request received: 57602 0022a300002ca0da
2023-09-02T01:37:29.134Z zigbee-herdsman:adapter:ezsp:debg Requesting 'Node Descriptor' for '57602'
2023-09-02T01:37:29.153Z zigbee-herdsman:adapter:ezsp:debg processMessage: {"messageType":0,"apsFrame":{"profileId":260,"clusterId":2820,"sourceEndpoint":1,"destinationEndpoint":1,"options":64,"groupId":0,"sequence":221},"lqi":136,"rssi":-66,"sender":33556,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[24,43,10,11,5,41,116,2]}}
2023-09-02T01:37:29.157Z zigbee-herdsman:adapter:ezsp:debg processMessage: {"messageType":0,"apsFrame":{"profileId":260,"clusterId":2820,"sourceEndpoint":1,"destinationEndpoint":1,"options":64,"groupId":0,"sequence":206},"lqi":156,"rssi":-61,"sender":8320,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[24,15,10,11,5,41,2,0]}}
2023-09-02T01:37:29.195Z zigbee-herdsman:adapter:ezsp:debg processMessage: {"messageType":4,"apsFrame":{"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":74},"lqi":160,"rssi":-60,"sender":63352,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[99,253,255,4,1,1,25,0,0]}}
2023-09-02T01:37:29.607Z zigbee-herdsman:adapter:ezsp:debg processMessage: {"messageType":0,"apsFrame":{"profileId":260,"clusterId":1794,"sourceEndpoint":1,"destinationEndpoint":1,"options":320,"groupId":0,"sequence":72},"lqi":156,"rssi":-61,"sender":10857,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[8,19,10,0,0,37,198,43,0,0,0,0]}}
2023-09-02T01:37:29.612Z zigbee-herdsman:adapter:ezsp:debg sendZclFrameToEndpointInternal 0xb0ce1814001d7a6a:10857/1 (0,0,1)
Error: Failure send findKeyTableEntry:{"type":"Buffer","data":[113,0,1,117,0,218,160,44,0,0,163,34,0,1]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:562:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
```

An interview that keeps crashing things

Speedy1496, 2023-09-26:
Hi,
I have the same trying to pair Danfoss 14G2461
Most of time when I try to pair, the Add On restart
When I have the time to read the log, I found the same error
( Sonoff Dongle-E  6.10.3 firmware)

Thanks

franssonjohan, 2023-09-27:
Still having the same issue. Every other time the HA Addon crashed upon pairing a device, irrespective what kind of device I'm paring. Keeping the addon updated, now with 1.33.0

miquelbotanch, 2024-01-06:
In the same boat when trying to add a Aqara Human sensor. Zigbee2mqtt 1.34.0-dev

fastlane086, 2024-01-18:
The error is still present in version 1.35.1.

`2024-01-18T06:56:09.219Z zigbee-herdsman:adapter:ezsp:debg processMessage: {"messageType":0,"apsFrame":{"profileId":260,"sequence":116,"clusterId":2820,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":320},"lqi":116,"rssi":-71,"sender":28229,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[24,201,10,8,5,33,202,0]}}
2024-01-18T06:56:09.227Z zigbee-herdsman:adapter:ezsp:debg processMessage: {"messageType":0,"apsFrame":{"profileId":260,"sequence":36,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"lqi":144,"rssi":-64,"sender":4206,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[8,169,10,0,0,16,0,1,64,33,0,0,2,64,33,0,0,2,128,48,0]}}
2024-01-18T06:56:09.411Z zigbee-herdsman:adapter:ezsp:debg processMessage: {"messageType":0,"apsFrame":{"profileId":260,"sequence":37,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"lqi":148,"rssi":-63,"sender":4206,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[8,169,10,0,0,16,0,1,64,33,0,0,2,64,33,0,0,2,128,48,0]}}
Error: Failure send findKeyTableEntry:{"type":"Buffer","data":[12,0,1,117,0,170,87,199,88,13,56,193,164,1]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:562:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)`

yoranschrootfancom, 2024-02-04:
I also still have this issue in version 1.35.2

Coordinator type: EZSP v12
Coordinator revision: 7.3.1.0 build 176

```
zigbee2mqtt  | Error: Failure send findKeyTableEntry:{"type":"Buffer","data":[67,0,1,117,0,26,100,229,0,1,136,23,0,1]}
zigbee2mqtt  |     at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:646:23
zigbee2mqtt  |     at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
zigbee2mqtt  |     at Driver.cleanupTClinkKey (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:419:24)
zigbee2mqtt exited with code 0
```