Koenkk/zigbee2mqtt #17533: Cannot remove device from group

corporategoth, 2023-05-02:
### What happened?

I have a group that includes a device
I can't remove that device from the group.  No matter what I try.

### What did you expect to happen?

I click the trash can, the entity is removed from the group

### How to reproduce it (minimal and precise)

Have an entry in the group.
Try and remove it.

https://user-images.githubusercontent.com/3732622/235605305-698c565f-80aa-4014-89ec-e8a312ac0101.mp4

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

6.10.3.0 build 297

### Adapter

ezsp

corporategoth, 2023-06-02:
Uhh, any update on this ticket?  It's being allowed to grow stale as it's not being responded to.

fumantsu, 2023-07-08:
up

fumantsu, 2023-08-09:
Not stale yet

Send from my hotdog

fumantsu, 2023-09-09:
Not stale

Send from my hotdog

closed 2023-10-18 by github-actions[bot]

Stefkeys, 2024-01-04:
Did you ever find an answer to this? 

I'm unable to remove any devices period. Not on its own or from a group. Getting Mac address failure even though I've made sure I don't have zigbee and wifi on the same channel and I've unplugged the dongle and plugged it back in. Rebooted Z2M. And rebooted HA.

It's frustrating that valid issues get closed out just because people don't have enough time to look into it right away.