Koenkk/zigbee2mqtt #17534: No connection since 1.30.4 update

Chacsam, 2023-05-02:
### What happened?

Since update to 1.30.4, my container keeps crashing.

The error I receive is
```
Zigbee2MQTT:info  2023-05-02 10:21:28: Zigbee2MQTT started!
Error: connect ECONNREFUSED 85.214.195.163:9005
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1494:16)
```
What's that external IP it's trying to connect to?

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

20221226

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

Chacsam, 2023-05-03:
OK... it is somehow solved now. A night doing nothing did help

closed 2023-05-03 by Chacsam

AugustinN, 2023-05-03:
I have the exact same issue (same ip adress), and doing nothing has not solved it...anyone has a clue what might be going on ? Running zigbee2mqtt on an old raspberry pi b+

Chacsam, 2023-05-03:
It's good to feel less lonely. 
As a rollback didn't work, I assumed it wasn't linked to the update so I closed this issue to have a more 'to the point' description in #17550

Palhersan, 2023-08-04:
Solucionado:
https://www.reddit.com/r/Zigbee2MQTT/comments/137c9ar/zigbee2mqtt_connecting_to_external_server/