Koenkk/zigbee2mqtt #17535: No Connection to Conbee with Docker

boscho87, 2023-05-02:
### What happened?

I Updated my Raspberry PI and restartet the docker instance

### What did you expect to happen?

Should Connect Like before the Update

### How to reproduce it (minimal and precise)

Its Possible that i have the Same Problem with my zigbee2mqtt docker configuraiton?

```shell
zigbee2mqtt  | Zigbee2MQTT:error 2023-05-02 08:34:59: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
zigbee2mqtt  |     at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
zigbee2mqtt  |     at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
zigbee2mqtt  |     at Zigbee.start (/app/lib/zigbee.ts:58:27)
zigbee2mqtt  |     at Controller.start (/app/lib/controller.ts:101:27)
zigbee2mqtt  |     at start (/app/index.js:107:5)
```

```yaml
 zigbee2mqtt:
    container_name: zigbee2mqtt
    restart: always
    image: koenkk/zigbee2mqtt:1.28.4
    volumes:
      - ./zigbee2mqtt-data:/app/data
      - /run/udev:/run/udev:ro
    ports:
      - 8080:8080
    environment:
      - TZ=Europe/Zurich
    devices:
      - /dev/ttyACM0:/dev/ttyACM0
```

```shell
[ 1078.028698] usb 1-1.3: USB disconnect, device number 18
[ 1078.254314] usb 1-1.3: new full-speed USB device number 19 using xhci_hcd
[ 1078.363336] usb 1-1.3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[ 1078.363350] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1078.363356] usb 1-1.3: Product: ConBee II
[ 1078.363361] usb 1-1.3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[ 1078.363366] usb 1-1.3: SerialNumber: DE2662453
[ 1078.368759] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
```

my devices info:
```shell
docker@dockerpi:~ $ ls -lah /dev/ttyACM0 
crw-rw---- 1 root dialout 166, 0 May  2 10:18 /dev/ttyACM0

docker@dockerpi:~ $ ls -la /dev/serial/by-path/
total 0
drwxr-xr-x 2 root root 60 May  2 10:18 .
drwxr-xr-x 3 root root 60 May  2 10:18 ..
lrwxrwxrwx 1 root root 13 May  2 10:18 platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.3:1.0 -> ../../ttyACM0
```

anyone knows why i dont have an `/dev/serial/by-id` path?

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

-

### Adapter

-

zedocrob, 2023-05-02:
hi, same problem after last docker container update and restarting docker  with SONOFF Zigbee 3.0 USB Dongle Plus V2.
`/dev/serial/by-id` not found. 
container not starting.

My device is here
```
root@nas:~# ls -lah /dev/ttyACM0 
crw-rw---- 1 root dialout 166, 0  2 mai   17:10 /dev/ttyACM0
root@nas:~# ls -la /dev/serial/by-path/
total 0
drwxr-xr-x 2 root root 60  2 mai   17:10 .
drwxr-xr-x 3 root root 60  2 mai   17:10 ..
lrwxrwxrwx 1 root root 13  2 mai   17:10 pci-0000:00:1d.0-usb-0:1.7:1.0 -> ../../ttyACM0
```

passibe15, 2023-05-02:
See [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6937#issuecomment-1529697409) and [here](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1035094), this seems to be an issue with udev.

It is fixed in udev (or systemd? I am not sure …) version 252.5-2~bpo11+1. I updated both, systemd and udev from bullseye-backports, now everything is working fine.

Steps:
1. Create `/etc/apt/sources.list.d/bullseye-backports.list` containing
```
# bullseye-backports
deb http://deb.debian.org/debian bullseye-backports main
```

2. Run `sudo apt update`
3. Run `sudo apt-get -t bullseye-backports install systemd udev`

I threw in a reboot and then everything started up again. Of course, you have make a mental note that systemd and udev are installed from backports now …

boscho87, 2023-05-03:
> See [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6937#issuecomment-1529697409) and [here](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1035094), this seems to be an [...]

Nice, works also for me. whats the difference between bullseye-backports and the "common" packages?

ismoo1, 2023-05-29:
I created the file run the commands but did not help in RPI4:

i@raspberrypi:~ $ sudo apt update
Get:1 http://deb.debian.org/debian bullseye-backports InRelease [49.0 kB]
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:3 http://phoscon.de/apt/deconz bullseye InRelease
Get:4 https://packages.grafana.com/oss/deb stable InRelease [5,984 B]
Hit:5 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:6 https://deb.nodesource.com/node_16.x bullseye InRelease
Get:7 https://repos.influxdata.com/debian bullseye InRelease [7,040 B]
Get:8 https://download.docker.com/linux/raspbian bullseye InRelease [26.7 kB]
Err:1 http://deb.debian.org/debian bullseye-backports InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 648ACFD622F3D138 NO_PUBKEY 0E98404D386FA1D9
Err:4 https://packages.grafana.com/oss/deb stable InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 9E439B102CF3C0C6
Err:7 https://repos.influxdata.com/debian bullseye InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY D8FF8E1F7DF8B07E
Reading package lists... Done
W: GPG error: http://deb.debian.org/debian bullseye-backports InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 648ACFD622F3D138 NO_PUBKEY 0E98404D386FA1D9
E: The repository 'http://deb.debian.org/debian bullseye-backports InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://packages.grafana.com/oss/deb stable InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 9E439B102CF3C0C6
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://repos.influxdata.com/debian bullseye InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY D8FF8E1F7DF8B07E

pi@raspberrypi:~ $ sudo apt-get -t bullseye-backports install systemd udev
Reading package lists... Done
E: The value 'bullseye-backports' is invalid for APT::Default-Release as such a release is not available in the sources

My OS is
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

closed 2023-07-06 by github-actions[bot]