Koenkk/zigbee2mqtt #17537: [Wrong device]: UK TuYa 20A Zigbee Smart Plug variant picture

busywait, 2023-05-02:
### Link

https://www.aliexpress.com/item/1005004945906794.html

### Model

TS011F_plug_3

### Description

UK smart plug (with power monitoring by polling)

### Vendor

TuYa

### Picture (link)

https://ae01.alicdn.com/kf/S8926bd65884e4868952bfad48622a7b9o/Tuya-20A-Zigbee-Smart-Plug-UK-Socket-Power-Strip-Mini-Home-Wireless-Outlet-with-Energy-Monitor.jpg_640x640.jpg

### Database entry

{"id":36,"type":"Router","ieeeAddr":"0x70b3d52b60023aed","nwkAddr":20533,"manufId":4660,"manufName":"_TZ3000_okaz9tjs","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,6,3,4,5,57345,2820,1794],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_okaz9tjs","powerSource":1,"zclVersion":2,"appVersion":100,"stackVersion":1,"hwVersion":2,"dateCode":"20210625","swBuildId":"500+TZSKT31BS102"}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":247,"rmsCurrent":10,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,0]}},"genOnOff":{"attributes":{"onOff":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":100,"stackVersion":1,"hwVersion":2,"dateCode":"20210625","swBuildId":"500+TZSKT31BS102","zclVersion":2,"interviewCompleted":true,"meta":{"configured":2123427102},"lastSeen":1683023700183,"defaultSendRequestWhen":"immediate"}

### Notes

- I bought this from the KAIYE Store/Ali Express page and it arrived in very generic packaging not even specific to this model. I expect many different sources will sell identical switches, I've stated "TuYa". 
- The UK model looks very different to the current picture in the DB, I've linked to a vendor image that can be cropped. 
- I added the text "UK" to the current description.

Koenkk, 2023-05-02:
The model cannot be the same as the existing model (`TS011F_plug_3`), is there any model written on the device?

Koenkk, 2023-05-03:
Thanks! Will be fixed with the next release.

closed 2023-05-03 by Koenkk