Koenkk/zigbee2mqtt #17539: [Wrong device]:

mbrackjr, 2023-05-02:
### Link

https://www.zemismart.com/products/zn-lc1e

### Model

ZN-LC1E

### Description

Smart Curtain/Shutter Switch

### Vendor

Zemismart

### Picture (link)

https://s.alicdn.com/@sc04/kf/H1d172881c39c41589583881cf1ebde52D.jpg_960x960.jpg

### Database entry

{"id":33,"type":"Router","ieeeAddr":"0xcc86ecfffe56b5ef","nwkAddr":37729,"manufId":4098,"manufName":"_TZ3000_74hsp7qy","powerSource":"Mains (single phase)","modelId":"TS130F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,4,5,6,258],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS130F","manufacturerName":"_TZ3000_74hsp7qy","stackVersion":0,"dateCode":"","zclVersion":3}},"genOnOff":{"attributes":{"tuyaBacklightMode":0}},"closuresWindowCovering":{"attributes":{"32768":0,"currentPositionLiftPercentage":84,"tuyaMovingState":1,"tuyaCalibration":1,"tuyaMotorReversal":0,"moesCalibrationTime":302}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021a8eb44","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1683031405535,"defaultSendRequestWhen":"immediate"}

### Notes

Functionally identical to Tuya TS207, but Zemismart buttons have arrows (up and down) better suited for shutters

Koenkk, 2023-05-02:
Thanks, will be fixed with the next release.

closed 2023-05-02 by Koenkk

pereiraru, 2024-10-08:
hi there, 

just installed the Zemismart ZN-LC1E, but it's being recognised in zigbee2mqtt as tze204_ettzhnwk . Latest stable version of zigbee2mqtt to date.