Koenkk/zigbee2mqtt #17540: [Wrong device]: Picture

salopette, 2023-05-02:
### Link

https://de.aliexpress.com/item/1005005175597153.html?pdp_npi=2%40dis%21EUR%21%E2%82%AC%2046.12%21%E2%82%AC%2025.36%21%21%21%21%21%40211b612816830399453507565ec937%2112000031972701604%21btf&_t=pvid%3Ad76401e2-d607-4231-8f38-46c61c944c3c&afTraceInfo=1005005175597153__pc__pcBridgePPC__xxxxxx__1683039945&spm=a2g0o.ppclist.product.mainProduct&gatewayAdapt=glo2deu

### Model

TS0601

### Description

Human Presence Detector

### Vendor

Tuya

### Picture (link)

https://zigbee.blakadder.com/assets/images/devices/Tuya_ZY-M100-S.webp

### Database entry

{"id":30,"type":"Router","ieeeAddr":"0x187a3efffe0f43eb","nwkAddr":39861,"manufId":4098,"manufName":"_TZE204_ztc6ggyl","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"���+i","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_ztc6ggyl","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1683037793298,"defaultSendRequestWhen":"immediate"}

### Notes

The bone has the wrong picture.

Koenkk, 2023-05-02:
`TS0601` is the generic model number, please provide the model number listed on the device.

salopette, 2023-05-02:
[image]

Koenkk, 2023-05-03:
The zigbee Manufacturer name is not what I'm searching for. I'm looking for a model number that is written on the device itself.

salopette, 2023-05-03:
[image]
[image]
[image]

Koenkk, 2023-05-04:
I see now that this has already been addressed in https://github.com/Koenkk/zigbee2mqtt/issues/17525, fix will be included in the next release.

closed 2023-05-04 by Koenkk

salopette, 2023-06-03:
Hello, on 1.6 with the update the right picture was there. now it is again a different picture than built-in version.