Koenkk/zigbee2mqtt #17541: Develco SIRZB-110 mode and level mismatch

singloon, 2023-05-02:
### What happened?

Develco SIRZB-110 observed from 1.30.3 and 1.30.4-dev commit. 

The level and modes are not configuring the correct properties.

- It seems the level is configuring the mode.
-  And the modes is configuring the level (volume).

I haven't been able to figure out what else may not be mapped accordingly.

### What did you expect to happen?

When I set level and mode to be reflected

### How to reproduce it (minimal and precise)

Set the mode and level of device.
Trigger alarm.

### Zigbee2MQTT version

1.30.4-dev commit

### Adapter firmware version

20220219

### Adapter

slaesh's CC2652RB

bonjour81, 2023-06-05:
Hello @singloon, did you solve the issue?

singloon, 2023-06-05:
> Hello @singloon, did you solve the issue?

No solution

closed 2023-07-14 by github-actions[bot]