Koenkk/zigbee2mqtt #17542: After update to  1.30.4: TS011F plugs report wrong power (also shows Error)

snooopy365, 2023-05-02:
### What happened?

After doing the update, HA did show constant power consumtion on all TS011F plugs in my house.
I looked into Zigbee2MQTT and it does show the same high values.

I did delete all TS011F devices and added them again. It did not solve the problem.
Some plugs keep showing the highest power that was used constantly. Other ones do lower the number but still way about 0 even if the devices  are all off.
Others don't report any power at all and are just reporting "null".

The Error reported is: "Exception while calling from Zigbee converter: Cannot read properties of undefined (reading 'power')}"

### What did you expect to happen?

I did expect it to work like before.

### How to reproduce it (minimal and precise)

Update to 1.30.4

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

20230405

### Adapter

zStack3x0 (sonoff)

### Debug log

info  2023-05-02 14:16:42: MQTT publish: topic 'zigbee2mqtt/Steckdose Couch', payload '{"child_lock":null,"current":0.15,"energy":5.39,"indicator_mode":null,"linkquality":83,"power":14,"power_outage_memory":null,"state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":229}'
info  2023-05-02 14:16:42: MQTT publish: topic 'zigbee2mqtt/Steckdose Couch', payload '{"child_lock":null,"current":0.12,"energy":5.39,"indicator_mode":null,"linkquality":87,"power":10,"power_outage_memory":null,"state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":229}'
info  2023-05-02 14:16:44: MQTT publish: topic 'zigbee2mqtt/Thermometer Einlieger', payload '{"battery":100,"humidity":59,"linkquality":83,"temperature":16.6,"voltage":3000}'
info  2023-05-02 14:16:47: MQTT publish: topic 'zigbee2mqtt/Spülmaschine', payload '{"child_lock":"UNLOCK","current":0,"energy":25.01,"indicator_mode":"off/on","linkquality":83,"power":0,"power_outage_memory":"restore","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":230}'
info  2023-05-02 14:16:47: MQTT publish: topic 'zigbee2mqtt/Spülmaschine', payload '{"child_lock":"UNLOCK","current":0,"energy":25.01,"indicator_mode":"off/on","linkquality":83,"power":0,"power_outage_memory":"restore","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":227}'
info  2023-05-02 14:16:47: MQTT publish: topic 'zigbee2mqtt/Steckdose Couch', payload '{"child_lock":null,"current":0.18,"energy":5.39,"indicator_mode":null,"linkquality":83,"power":10,"power_outage_memory":null,"state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":229}'
info  2023-05-02 14:16:52: MQTT publish: topic 'zigbee2mqtt/Steckdose Couch', payload '{"child_lock":null,"current":0.12,"energy":5.39,"indicator_mode":null,"linkquality":83,"power":10,"power_outage_memory":null,"state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":229}'
info  2023-05-02 14:16:54: MQTT publish: topic 'zigbee2mqtt/Thermometer Einlieger', payload '{"battery":100,"humidity":59,"linkquality":80,"temperature":16.9,"voltage":3000}'
info  2023-05-02 14:16:57: MQTT publish: topic 'zigbee2mqtt/Steckdose Heizungsraum', payload '{"child_lock":"UNLOCK","current":0,"energy":0.1,"indicator_mode":"off/on","linkquality":80,"power":0,"power_outage_memory":"restore","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":230}'
info  2023-05-02 14:16:58: MQTT publish: topic 'zigbee2mqtt/Steckdose Ankleideraum', payload '{"child_lock":"UNLOCK","current":0,"energy":5.31,"indicator_mode":"off/on","linkquality":83,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":230}'
info  2023-05-02 14:16:58: MQTT publish: topic 'zigbee2mqtt/Steckdose Ankleideraum', payload '{"child_lock":"UNLOCK","current":0,"energy":5.31,"indicator_mode":"off/on","linkquality":83,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":228}'
error 2023-05-02 14:16:59: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'power')}
error 2023-05-02 14:16:59: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'power')}
error 2023-05-02 14:16:59: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'power')}
error 2023-05-02 14:16:59: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'power')}
error 2023-05-02 14:17:00: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'power')}
error 2023-05-02 14:17:00: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'power')}
error 2023-05-02 14:17:00: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'power')}
error 2023-05-02 14:17:00: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'power')}
error 2023-05-02 14:17:01: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'power')}
info  2023-05-02 14:17:01: MQTT publish: topic 'zigbee2mqtt/Steckdose Heizungsraum', payload '{"child_lock":"UNLOCK","current":0,"energy":0.1,"indicator_mode":"off/on","linkquality":0,"power":0,"power_outage_memory":"restore","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":229}'
error 2023-05-02 14:17:01: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'power')}

Koenkk, 2023-05-02:
I expect this is the same as #16709

Koenkk, 2023-05-03:
Published a possible fix in https://github.com/Koenkk/zigbee2mqtt/issues/16709#issuecomment-1533483208, let's continue there.

closed 2023-05-03 by Koenkk