Koenkk/zigbee2mqtt #17548: adapter Failed to validate path: 'Error: spawn udevadm ENOENT

cloudbr34k84, 2023-05-02:
### What happened?

I was adding some more devices and I pressed the join button and then pressed it again to stop the join process when I pressed it again, I noticed that the timer was not working and I started to get errors.  
Now the adapter won't stay connected and I'm getting the above error in the logs as well as Error: SRSP - SYS - getExtAddr after 6000ms
This happens when  I reach a larger number of devices. I was on 97 when this happened. The firmware just starts to slow down and then it cant handle things for some reason.

### What did you expect to happen?

I expected that I could add devices without this error continuously happening

### How to reproduce it (minimal and precise)

I was adding some more devices and I pressed the join button and then pressed it again to stop the join process when I pressed it again, I noticed that the timer was not working and I started to get errors.  

### Zigbee2MQTT version

1.30.4-1

### Adapter firmware version

20230410

### Adapter

Zigbee ESP32GW (Zigstar)

### Debug log

Multiple logs, hope this helps. 

[2023-05-02T200538.172Z zigbee-herds.txt](https://github.com/Koenkk/zigbee2mqtt/files/11377377/2023-05-02T200538.172Z.zigbee-herds.txt)
[log (1).txt](https://github.com/Koenkk/zigbee2mqtt/files/11377378/log.1.txt)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/11377379/log.txt)

cloudbr34k84, 2023-05-03:
@Koenkk i think i found out one of the reason the firmware throws the (SRSP - AF - dataRequest after 6000ms). It does not like it when you have multiple devices trying to pair. I had 5 lights that I had reset and came up pairing. i tried to stop it but the damage had been done..

cloudbr34k84, 2023-05-04:
@Koenkk a follow up to this is I'm now at 116 devices and I have noticed Z@m gets a lot slower, and on occasion when I press the Permit Join button nothing happens and times out
I have seen this error
```
debug 04-05-2023 20:14:26: Error: Command 0x70ac08fffe916c19/1 manuSpecificTuya.mcuGatewayConnectionStatus({"payloadSize":1,"payload":1}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)
```

cloudbr34k84, 2023-05-04:
Another update
Im reverting to the previous firmware to see if that's any better.

Koenkk, 2023-05-04:
Let me know how that firmware works out. I believe your coordinator runs out of memory, this is probably due to a lot of messages in your zigbee network (the TuYa devices are spamming).

- I will provide a firmware with more memory later, keep an eye on https://github.com/Koenkk/Z-Stack-firmware/issues/439
- Can you provide the herdsman debug log starting from the point where you start z2m until this error? (the log you provided is really short)

cloudbr34k84, 2023-05-04:
Yeah I have these tuya 6 in 1 devices and they constantly report, it nuts  It would be great to change the reporting on these. 

Regarding the move back down to previous firmware,  I gave woke up to all my devices online.  The real test is when I try to add some more devices today. 

I will try and see if I can break it,  as I now have a way to restore. 

Btw these hamgeek poe adapters are great!  Easy to update firmware,  easy to restart if I see the 6000 error.

cloudbr34k84, 2023-05-04:
> Let me know how that firmware works out. I believe your coordinator runs out of memory, this is probably due to a lot of messages in your zigbee network (the TuYa devices are spamming).
> 
> * I will provide a firmware with more memory later, keep an eye on [Z-Stack_3.x.0 coordinator 20230410 feedback Z-Stack-firmware#439](https://github.com/Koenkk/Z-Stack-firmware/issues/439)
> * Can you provide the herdsman debug log starting from the point where you start z2m until this error? (the log you provided is really short)

Hey @Koenkk here is a log with the error. It happened when I pressed Permit Join, but then after a min, it appears the Permit Join worked as the timer started. but still.
I have so many messages for my devices that end with word Climate..
[log (2).txt](https://github.com/Koenkk/zigbee2mqtt/files/11402154/log.2.txt)

Koenkk, 2023-05-06:
This is not the herdsman debug logging, 
See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

closed 2023-05-17 by cloudbr34k84