Koenkk/zigbee2mqtt #17549: [New device support]:  _TZE204_6wi2mope

jonas59garcia, 2023-05-03:
### Link

https://www.aliexpress.com/item/1005001622518790.html?spm=a2g0o.order_list.order_list_main.16.510d369699tzWo&gatewayAdapt=glo2ita

### Database entry

{"id":10,"type":"Router","ieeeAddr":"0xa4c13858bf7aba6e","nwkAddr":9621,"manufId":4417,"manufName":"_TZE204_6wi2mope","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�\t�+i�\t�+i�\t�+i�\t�+i","65506":56,"65508":0,"appVersion":74,"modelId":"TS0601","manufacturerName":"_TZE204_6wi2mope","powerSource":1,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1682802134730,"defaultSendRequestWhen":"immediate"}

### Comments

model:  TS0601 
Zigbee Manufacturer: _TZE204_6wi2mope
Support status:  Not supported

Light Switch 4 gang 

I didn't have the necessary knowledge to add the device

Koenkk, 2023-05-03:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_aqnazj70', '_TZE204_6wi2mope', '_TZE200_k6jhsr0q', '_TZE200_di3tfv5b', '_TZE200_mexisfik']),
    model: 'TS0601_switch_4_gang',
    vendor: 'TuYa',
    description: '4 gang switch',
    exposes: [e.switch().withEndpoint('l1').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l2').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l3').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l4').setAccess('state', ea.STATE_SET)],
    fromZigbee: [fz.ignore_basic_report, fz.tuya_switch],
    toZigbee: [tz.tuya_switch_state],
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        // Endpoint selection is made in tuya_switch_state
        return {'l1': 1, 'l2': 1, 'l3': 1, 'l4': 1};
    },
};

module.exports = definition;

```
- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

jonas59garcia, 2023-05-04:
it worked, thank you

Koenkk, 2023-05-04:
Thanks, you can remove the ext converter after updating to the next release.

closed 2023-05-04 by Koenkk