Koenkk/zigbee2mqtt #17550: Crash due to connection to external IP (Strato AG???)

Chacsam, 2023-05-03:
### What happened?

Since +/- 48 hours, Zigbee2mqtt is randomly crashing on the following error:
```
Error: connect ECONNREFUSED 85.214.195.163:9005
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1494:16)
```

**Note**: I only see this error in the _Docker container log_, not in the Zigbee2mqtt log. (attached)

Basically the container starts up correctly, connects to the mosquitto eclipse broker as it should, I even see the values being updated in my Home Assistant, but it crashes after 5 to 10 seconds with the above mentioned error.

Sometimes, it doesn't crash and stays stable, but a restart might get it in crashing mode again.

The lookup of that IP refers to some German Provider called Strato, which I have no contact with.
Does anyone know where it comes from? Zigbee2mqtt? One of my devices?

### What did you expect to happen?

Zigbee2mqtt to stay stable

### How to reproduce it (minimal and precise)

It looks random on start

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

20221226

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/11380244/log.txt)

liakos356, 2023-05-04:
I have the same issue with Conbee 2 adapter

albertgaes, 2023-05-04:
I have the same issue. Mine is pointing another port:

Error: connect ECONNREFUSED 85.214.195.163:9007
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1494:16)

Sonoff Dongle 3.0

Chacsam, 2023-05-04:
Surfing to that IP on normal http port is actually pretty surprising...

Koenkk, 2023-05-04:
can someone provide the herdsman debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

liakos356, 2023-05-04:
> Yes, the exact same IP and port. I'm suspecting that the culprit is my connection that is behind a pi-hole. When I return from work I'll check if there is a new rule that blocks the connection there.

Removed pi-hole filtering. No change. Problem persists.

Chacsam, 2023-05-05:
> can someone provide the herdsman debug log of this?
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

@Koenkk Is this what you are looking for?
[log.zip](https://github.com/Koenkk/zigbee2mqtt/files/11403721/log.zip)

liakos356, 2023-05-05:
The content on this IP is like the personal file storage of some individual. Movies and series. What can possibly points to that?

liakos356, 2023-05-05:
I think I've figured it out. In the `./zigbee2mqtt/extension` folder was an `example-user-extension1682976504061.js` file that was making the call to this questionable server. When I rename the file to .bak, everything came back to normal. 

I think that the file was created by me when I wanted to add support to a Chinese no name energy clamp meter. The server url was inside the file contents, and it might point to the personal server of the person that made the guide I've followed. Nevertheless, despite the fix, the clamp meter is still functional.

albertgaes, 2023-05-05:
> I think I've figured it out. In the `./zigbee2mqtt/extension` folder was an `example-user-extension1682976504061.js` file that was making the call to this questionable server. When I rename the file [...]

Great! I renamed that file and now is working. Thanks a lot.

Chacsam, 2023-05-05:
> I think I've figured it out. In the `./zigbee2mqtt/extension` folder was an `example-user-extension1682976504061.js` file that was making the call to this questionable server. When I rename the file [...]

That solved the issue for me as well. Thanks

liakos356, 2023-05-05:
> One question though, @liakos356 , how would that test file created by you end up on my instance?

I can only guess that you might have followed the same guide that included the creation of the file, or that there were files included in that guide and you copied and paste them. Have you ever needed support for a Zigbee device that didn't recognized directly from the UI of the zigbee2mqtt?

liakos356, 2023-05-05:
> Never, I only have paired devices through the UI (and I only have mainstream devices: Sonoff, Philips, Ikea & Lidl). Also checked my drive and that file was modified on May 2 at 2.01am, which matches the moment when Watchtower updated my Zigbee2mqtt to version 1.30.4...

So your initial suspicion that the update broke the functionality is correct. And when someone downgraded to previous version, the file was still present and broke the installation. The real mystery is the type of content of this ip.

Chacsam, 2023-05-05:
> The real mystery is the type of content in this ip.

Fully agree. It's still a intriguing question.

liakos356, 2023-05-05:
Follow up : 
The file seems to be created in the following menu of the UI :

[image]

[image]

Koenkk, 2023-05-05:
Great that you found the issue, lesson learned: don't post random external extension from the internet 😄

closed 2023-05-05 by Koenkk

taimaiduc, 2023-05-13:
> I think I've figured it out. In the `./zigbee2mqtt/extension` folder was an `example-user-extension1682976504061.js` file that was making the call to this questionable server. When I rename the file [...]

Thank you for saving my time