Koenkk/zigbee2mqtt #17552: Payload is not supported (e.g. open, closed, opening, closing, stopped): STOP

KitHubek, 2023-05-03:
### What happened?

Error at log as below.

Mosquitto Current version: 6.2.1
Zigbee2MQTT Current version: 1.30.4-1
as i remember version previous no error

### What did you expect to happen?

no errror

### How to reproduce it (minimal and precise)

have device: AM43-0.45/40-ES-EB or AQARA ZNJLBL01LM

### Zigbee2MQTT version

1.30.4-1

### Adapter firmware version

0x26780700

### Adapter

ConBee2/RaspBee2

### Debug log

Logger: homeassistant.components.mqtt.cover
Source: components/mqtt/cover.py:395
Integration: MQTT (documentation, issues)
First occurred: 2 maja 2023 19:59:31 (53 occurrences)
Last logged: 10:22:19

Payload is not supported (e.g. open, closed, opening, closing, stopped): STOP

payloads:

MOES:

Received 13:23:58
QoS: 0
Payload: factory_reset: null
last_seen: '2023-05-03T13:23:58+02:00'
linkquality: 255
motor_direction: null
motor_speed: 50
opening_mode: null
position: 33
running: false
set_bottom_limit: null
set_upper_limit: null
state: STOP

AQARA:

Received 13:24:20
QoS: 0
Payload: battery: 92
charging_status: false
device_temperature: 24
illuminance: 0
illuminance_lux: 0
last_seen: '2023-05-03T13:24:20+02:00'
linkquality: 255
motor_state: pause
position: 100
power_outage_count: 1
running: false
state: 'OFF'
temperature: 25
update:
  installed_version: -1
  latest_version: -1
  state: idle
update_available: false

anthfarr, 2023-05-05:
Have the same issue exactly

Idcsl, 2023-05-10:
I see the same problem too. I have 4x MOES AM43 blind controllers and all show the same issue in the logs.

solarssk, 2023-05-19:
Same here with AM43

**What did you expect to happen?**
no errror
**How to reproduce it (minimal and precise)**
AM43-0.45/40-ES-EB
**Zigbee2MQTT version**
1.30.4
**Adapter firmware version**
0x26720700
**Adapter**
ConBee2/RaspBee2
**Debug log**
Logger: homeassistant.components.mqtt.cover
Source: components/mqtt/cover.py:396 
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22)) 
First occurred: May 18, 2023 at 7:12:37 PM (20 occurrences) 
Last logged: 4:35:37 AM

Payload is not supported (e.g. open, closed, opening, closing, stopped): STOP

cokeman0, 2023-05-25:
Hope light sensor and battery level will be added also :-)

closed 2023-05-26 by Koenkk