Koenkk/zigbee2mqtt #17556: Zigbee switch controlling wrong device.

glennpierce, 2023-05-03:
### What happened?

I'm and not sure if this is a bug or not as the behaviour is so strange. 

I have a SONOFF ZigBee 3.0 stick.
4 Zigbee power outlets eWeLink | SA-003-Zigbee

and a aquara double switch recognised as Xiaomi | WXKG15LM

Now I had the the left button Aqara activate a wifi device and the right button do some TTS to my amazon Alexa (through node red)

(I didn't have a major use and was playing around entertaining the kids)

this was working for months fine. The one day it stopped responding and so I ending up removing the device from zigbee2mqtt and rediscovering by holding the buttons down for five seconds until the led's flashed.

This was the re-entered into zigbee2mqtt but now the buttons start activating my zigbee plugs ?

Even if I shut down nodered it does this. 

Whats going on here ?

Thanks  

### What did you expect to happen?

I expected after re initialising the switch the original behaviour occurred and does not associate to random devices (I wasn't aware it was possible). 

This may not be a bug and my lack of understanding.

### How to reproduce it (minimal and precise)

Remove my switch and rejoin it. Each time the results on what it controls seems to be random.

### Zigbee2MQTT version

1.30.3 commit: 24c6b2e

### Adapter firmware version

20210120

### Adapter

SONOFF ZigBee 3.0 stick

closed 2023-06-10 by github-actions[bot]

kirikorneev, 2023-07-15:
@glennpierce Have you managed to solve this problem? I have exactly the same behaviour with even 2 of these devices.

glennpierce, 2023-07-16:
> @glennpierce Have you managed to solve this problem? I have exactly the same behaviour with even 2 of these devices.

I had to literally remove everything and start joining stuff again. It is very odd. Still have no idea how how a switch can start controlling other devices. I assume no one else does either considering the closed status.

kirikorneev, 2023-07-17:
> I had to literally remove everything and start joining stuff again. It is very odd. Still have no idea how how a switch can start controlling other devices. I assume no one else does either [...]

I have upgraded to the very last version, 1.32.1 and suddenly it was working as it supposed to. It seems they somehow fixed the problem from previous 1.3x versions.

remkolems, 2024-01-04:
This has happened to me too with Xiaomi [WXKG15LM](https://www.zigbee2mqtt.io/devices/WXKG15LM.html#xiaomi-wxkg15lm). All my smart sockets in my home completely went off line and on again. 

Was working good, all of a sudden it completely went bogus ... speechless to be honest.

kirikorneev, 2024-01-08:
> This has happened to me too with Xiaomi [WXKG15LM](https://www.zigbee2mqtt.io/devices/WXKG15LM.html#xiaomi-wxkg15lm). All my smart sockets in my home completely went off line and on again.
> 
> Was working good, all of a sudden it completely went bogus ... speechless to be honest.

Which version was it? I am running 1.32.2 but with my previous experience I am not eager to update to 1.35.

001384, 2024-09-23:
Now I know that the button links to the routing device, and that routing device switches when the button is pressed。