Koenkk/zigbee2mqtt #17557: Pairing SNZB-02D fails

Rablet, 2023-05-03:
### What happened?

Pairing SNZB-02D keeps failing with an error:

`Info 2023-05-03 22:33:31Accepting joining not in blocklist device '0x30fb10fffe6ee503'
Info 2023-05-03 22:33:31Device '0x30fb10fffe6ee503' joined
Info 2023-05-03 22:33:31MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x30fb10fffe6ee503","ieee_address":"0x30fb10fffe6ee503"},"type":"device_joined"}'
Info 2023-05-03 22:33:31MQTT publish: topic 'homeassistant/sensor/0x30fb10fffe6ee503/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x30fb10fffe6ee503"],"manufacturer":"SONOFF","model":"Temperature and humidity sensor with screen (SNZB-02D)","name":"0x30fb10fffe6ee503","sw_version":"1.0.9"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"0x30fb10fffe6ee503 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x30fb10fffe6ee503","unique_id":"0x30fb10fffe6ee503_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Info 2023-05-03 22:33:31MQTT publish: topic 'homeassistant/sensor/0x30fb10fffe6ee503/temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x30fb10fffe6ee503"],"manufacturer":"SONOFF","model":"Temperature and humidity sensor with screen (SNZB-02D)","name":"0x30fb10fffe6ee503","sw_version":"1.0.9"},"device_class":"temperature","enabled_by_default":true,"name":"0x30fb10fffe6ee503 temperature","state_class":"measurement","state_topic":"zigbee2mqtt/0x30fb10fffe6ee503","unique_id":"0x30fb10fffe6ee503_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
Info 2023-05-03 22:33:31MQTT publish: topic 'homeassistant/sensor/0x30fb10fffe6ee503/humidity/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x30fb10fffe6ee503"],"manufacturer":"SONOFF","model":"Temperature and humidity sensor with screen (SNZB-02D)","name":"0x30fb10fffe6ee503","sw_version":"1.0.9"},"device_class":"humidity","enabled_by_default":true,"name":"0x30fb10fffe6ee503 humidity","state_class":"measurement","state_topic":"zigbee2mqtt/0x30fb10fffe6ee503","unique_id":"0x30fb10fffe6ee503_humidity_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.humidity }}"}'
Info 2023-05-03 22:33:31MQTT publish: topic 'homeassistant/sensor/0x30fb10fffe6ee503/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x30fb10fffe6ee503"],"manufacturer":"SONOFF","model":"Temperature and humidity sensor with screen (SNZB-02D)","name":"0x30fb10fffe6ee503","sw_version":"1.0.9"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0x30fb10fffe6ee503 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x30fb10fffe6ee503","unique_id":"0x30fb10fffe6ee503_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Info 2023-05-03 22:33:31MQTT publish: topic 'homeassistant/sensor/0x30fb10fffe6ee503/last_seen/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x30fb10fffe6ee503"],"manufacturer":"SONOFF","model":"Temperature and humidity sensor with screen (SNZB-02D)","name":"0x30fb10fffe6ee503","sw_version":"1.0.9"},"device_class":"timestamp","enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:clock","name":"0x30fb10fffe6ee503 last seen","state_topic":"zigbee2mqtt/0x30fb10fffe6ee503","unique_id":"0x30fb10fffe6ee503_last_seen_zigbee2mqtt","value_template":"{{ value_json.last_seen }}"}'
Info 2023-05-03 22:33:31MQTT publish: topic 'zigbee2mqtt/0x30fb10fffe6ee503', payload '{"battery":null,"device":{"applicationVersion":0,"dateCode":"20230105","friendlyName":"0x30fb10fffe6ee503","hardwareVersion":0,"ieeeAddr":"0x30fb10fffe6ee503","manufacturerID":4742,"manufacturerName":"SONOFF","model":"SNZB-02D","networkAddress":16602,"powerSource":"Unknown","softwareBuildID":"1.0.9","type":"EndDevice","zclVersion":8},"humidity":null,"last_seen":"2023-05-03T21:33:31.588Z","linkquality":98,"temperature":null}'
Info 2023-05-03 22:33:31Starting interview of '0x30fb10fffe6ee503'
Info 2023-05-03 22:33:31MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x30fb10fffe6ee503","ieee_address":"0x30fb10fffe6ee503","status":"started"},"type":"device_interview"}'
Info 2023-05-03 22:33:31Accepting joining not in blocklist device '0x30fb10fffe6ee503'
Info 2023-05-03 22:33:31MQTT publish: topic 'zigbee2mqtt/0x30fb10fffe6ee503', payload '{"battery":null,"device":{"applicationVersion":0,"dateCode":"20230105","friendlyName":"0x30fb10fffe6ee503","hardwareVersion":0,"ieeeAddr":"0x30fb10fffe6ee503","manufacturerID":4742,"manufacturerName":"SONOFF","model":"SNZB-02D","networkAddress":16602,"powerSource":"Unknown","softwareBuildID":"1.0.9","type":"EndDevice","zclVersion":8},"humidity":null,"last_seen":"2023-05-03T21:33:31.789Z","linkquality":98,"temperature":null}'
Debug 2023-05-03 22:33:31Received MQTT message on 'homeassistant/sensor/0x30fb10fffe6ee503/battery/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x30fb10fffe6ee503"],"manufacturer":"SONOFF","model":"Temperature and humidity sensor with screen (SNZB-02D)","name":"0x30fb10fffe6ee503","sw_version":"1.0.9"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"0x30fb10fffe6ee503 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x30fb10fffe6ee503","unique_id":"0x30fb10fffe6ee503_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Debug 2023-05-03 22:33:31Received MQTT message on 'homeassistant/sensor/0x30fb10fffe6ee503/temperature/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x30fb10fffe6ee503"],"manufacturer":"SONOFF","model":"Temperature and humidity sensor with screen (SNZB-02D)","name":"0x30fb10fffe6ee503","sw_version":"1.0.9"},"device_class":"temperature","enabled_by_default":true,"name":"0x30fb10fffe6ee503 temperature","state_class":"measurement","state_topic":"zigbee2mqtt/0x30fb10fffe6ee503","unique_id":"0x30fb10fffe6ee503_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
Debug 2023-05-03 22:33:31Received MQTT message on 'homeassistant/sensor/0x30fb10fffe6ee503/humidity/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x30fb10fffe6ee503"],"manufacturer":"SONOFF","model":"Temperature and humidity sensor with screen (SNZB-02D)","name":"0x30fb10fffe6ee503","sw_version":"1.0.9"},"device_class":"humidity","enabled_by_default":true,"name":"0x30fb10fffe6ee503 humidity","state_class":"measurement","state_topic":"zigbee2mqtt/0x30fb10fffe6ee503","unique_id":"0x30fb10fffe6ee503_humidity_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.humidity }}"}'
Debug 2023-05-03 22:33:31Received MQTT message on 'homeassistant/sensor/0x30fb10fffe6ee503/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x30fb10fffe6ee503"],"manufacturer":"SONOFF","model":"Temperature and humidity sensor with screen (SNZB-02D)","name":"0x30fb10fffe6ee503","sw_version":"1.0.9"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0x30fb10fffe6ee503 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x30fb10fffe6ee503","unique_id":"0x30fb10fffe6ee503_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Debug 2023-05-03 22:33:31Received MQTT message on 'homeassistant/sensor/0x30fb10fffe6ee503/last_seen/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device [... cut ...]
I've tried this well over 20 times this case as I've read the Sonoff devices can sometimes be tricky to pair (although my SNZB-02 and SNZB-01 devices have all been straightforward).

Is this likely to be a bug, faulty unit, or me doing something wrong?

### What did you expect to happen?

I expected it to pair successfully

### How to reproduce it (minimal and precise)

Push the pairing button for five seconds and wait

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

20221226

### Adapter

zzh!

closed 2023-06-10 by github-actions[bot]

sulliwane, 2023-07-28:
same issue here

sulliwane, 2023-07-28:
[image]
[image]
[image]

unsupported here :( any idea?