Koenkk/zigbee2mqtt #17559: [New device support]: _TZE204_atpwqgml

jonas59garcia, 2023-05-04:
### Link

https://pt.aliexpress.com/item/1005001622518790.html?spm=a2g0o.order_list.order_list_main.16.510d369699tzWo&gatewayAdapt=glo2bra

### Database entry

{"id":11,"type":"Router","ieeeAddr":"0xa4c138f32e222380","nwkAddr":19286,"manufId":4417,"manufName":"_TZE204_atpwqgml","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"���+i���+i���+i\u0000��+i;��+i;��+i<��+i<��+iw��+iw��+ix��+ix��+i","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1683169663689,"defaultSendRequestWhen":"immediate"}

### Comments

Zigbee Model: TS0601
Zigbee Manufacturer:   _TZE204_atpwqgml
Support status:   Not supported

Light Switch 3 gang

I didn't have the necessary knowledge to add the device

Koenkk, 2023-05-04:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_kyfqmmyl'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_2hf7x9n3'},
        {modelID: 'TS0601', manufacturerName: '_TZE204_atpwqgml'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_bynnczcb'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_atpwqgml'}],
    model: 'TS0601_switch_3_gang',
    vendor: 'TuYa',
    description: '3 gang switch',
    whiteLabel: [{vendor: 'NOVADIGITAL', model: 'WS-US-ZB', description: 'Interruptor touch Zigbee 3 Teclas'}],
    exposes: [e.switch().withEndpoint('l1').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l2').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l3').setAccess('state', ea.STATE_SET)],
    fromZigbee: [fz.ignore_basic_report, fz.tuya_switch],
    toZigbee: [tz.tuya_switch_state],
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
    },
    endpoint: (device) => {
        // Endpoint selection is made in tuya_switch_state
        return {'l1': 1, 'l2': 1, 'l3': 1};
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

jonas59garcia, 2023-05-05:
I already have this ext_converter.js file

How do I combine the 2 codes?

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_aqnazj70', '_TZE204_6wi2mope', '_TZE200_k6jhsr0q', '_TZE200_di3tfv5b', '_TZE200_mexisfik']),
    model: 'TS0601_switch_4_gang',
    vendor: 'TuYa',
    description: '4 gang switch',
    exposes: [e.switch().withEndpoint('l1').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l2').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l3').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l4').setAccess('state', ea.STATE_SET)],
    fromZigbee: [fz.ignore_basic_report, fz.tuya_switch],
    toZigbee: [tz.tuya_switch_state],
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        // Endpoint selection is made in tuya_switch_state
        return {'l1': 1, 'l2': 1, 'l3': 1, 'l4': 1};
    },
};

module.exports = definition;
```

Koenkk, 2023-05-05:
@jonas59garcia you can use different file names, e.g. `ext_converter_2.js`

jonas59garcia, 2023-05-06:
it's working, thank you.

Koenkk, 2023-05-06:
Good!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-05-06 by Koenkk