Koenkk/zigbee2mqtt #17560: [Wrong device]:

HittyStorm, 2023-05-04:
### Link

https://aliexpress.ru/item/4001051095315.html

### Model

ZN231392 or TS0111_valve

### Description

Smart water/gas valve

### Vendor

Tuya

### Picture (link)

https://www.turboimagehost.com/p/88480600/Valve.jpg.html

### Database entry

{"id":12,"type":"Router","ieeeAddr":"0x70b3d52b60024b80","nwkAddr":3741,"manufId":4660,"manufName":"_TZ3000_5ucujjts","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,6,3,4,5,57345],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"manufacturerName":"_TZ3000_5ucujjts","zclVersion":2,"appVersion":100,"modelId":"TS0001","powerSource":1}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00060fd6da","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":100,"stackVersion":1,"hwVersion":3,"dateCode":"20210625","swBuildId":"202+TZZSW31LZ103","zclVersion":2,"interviewCompleted":true,"meta":{"configured":153093895},"lastSeen":1683143300713,"defaultSendRequestWhen":"immediate"}

### Notes

Hello. I ordered a TuYa Zigbee 3.0 water/gas valve, but when pairing with Zigbee2Mqtt it's appears as [TS0001](https://www.zigbee2mqtt.io/devices/TS0001.html#tuya-ts0001) 1 gang switch.
Re-pairing is not helps me.
When I use on/off on tis switch — motor arm is moving, so device is function normal. But for now at least picture and description is wrong (and not only for me — people on AliExpress comments reports the same behavior). And it looks like power_on_behavior function is not working.

Koenkk, 2023-05-04:
Will be fixed with the next release, thanks for reporting.

closed 2023-05-04 by Koenkk

Aida3000, 2025-02-26:
It is the same with the device in the picture. 

[image]

Do you have a trick to solve the problem?
An identical device works fine:

[image]

vladopp, 2026-09-11:
I am facing the same issue - the valve actutaor (either ZN231392 or TS0111_valve) is recognized as [TS0001](https://www.zigbee2mqtt.io/devices/TS0001.html) 
I am on Zigbee2MQTT version: 2.14.1