Koenkk/zigbee2mqtt #17568: TS0222 by _TZ3000_kky16aay sensor [New device support]:

Dualmiracle, 2023-05-05:
### Link

https://nl.aliexpress.com/item/1005005218556209.html?

### Database entry

{"id":22,"type":"EndDevice","ieeeAddr":"0xa4c138393c571d40","nwkAddr":55611,"manufId":4417,"manufName":"_TZ3000_kky16aay","powerSource":"Battery","modelId":"TS0222","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":262,"inClusterList":[1,1026,1029,1024,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\"��+\u0012#��+\u0012","65506":54,"65508":0,"appVersion":66,"stackVersion":0,"dateCode":""}},"msTemperatureMeasurement":{"attributes":{"measuredValue":1500}},"msRelativeHumidity":{"attributes":{"measuredValue":680}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":166}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1683247066411,"defaultSendRequestWhen":"immediate"}

### Comments

Hi,

I want to add the above sensor. It's a sensor for Plants that measures humidity, temperature and light.

I get it into zigbee2mqtt but the problem is the humidity is 6,2% and it should by 62%. The standard humidity convertor devides is by 100 but it should be 10 with this sensor.

How can i fix that?

raphaabreu, 2023-05-12:
Hello @Dualmiracle, can you share the external converter you have used to set this up? I have the same device but I am not sure how to configure it.

Dualmiracle, 2023-05-13:
I did add the external converter. But as mentioned it doesnt function correct. the value is divided by 100 but I think it should be divided by 10 with this sensor but I don't know how to set that up.

raphaabreu, 2023-05-14:
I have the same sensor but I could not get it to report anything:

[image]

Dualmiracle, 2023-05-14:
> I have the same sensor but I could not get it to report anything:
> 
> [image]

Looks like you didnt set up an external converter. You can read here how that works: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

the data i put in the external converter is above this message.

Dualmiracle, 2023-05-14:
maybe @Koenkk could help us a little bit?

Dualmiracle, 2023-05-15:
> Could you check if the issue is fixed with the following external converter:
> 
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const utils = require('zigbee-herdsman-converters/lib/utils');
> const globalStore = require('zigbee-herdsman-converters/lib/store');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const fzLocal = {
>     humidity: {
>         ...fz.humidity,
>         convert: (model, msg, publish, options, meta) => {
>             const result = fz.humidity.convert(model, msg. publish, options, meta);
>             result.humidity *= 10;
>             return result;
>         },
>     }
> }
> const definition =  {
>     zigbeeModel: ['TS0222'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
>     model: 'TS0222_Soil', // Vendor model number, look on the device for a model number
>     vendor: 'TuYa', // Vendor of the device (only used for documentation and startup logging)
>     description: 'temperature & humidity sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
>     fromZigbee: [fzLocal.humidity, fz.battery, fz.temperature, fz.illuminance], // We will add this later
>     toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
>     exposes: [e.battery(), e.temperature(), e.humidity(), e.illuminance()],
>     };
> 
> module.exports = definition;
> ```
> 
> * save this as file next to `configuration.yaml` as `ext_converter.js`
> * add it to `configuration.yaml`:
> 
> ```yaml
> external_converters:
>   - ext_converter.js
> ```
> 
> * start z2m, check if issue is fixed

Unfortunatly its not working, it still gives 7% humidity and not 70%

raphaabreu, 2023-05-15:
Hi, 

In: 
```
const result = fz.humidity.convert(model, msg. publish, options, meta);
```

Is this really supposed to be `msg.publish`? I think it might have been a typo and a `,` should be there instead.

I have tried both ways but my unit still reports nothing.

Koenkk, 2023-05-15:
@raphaabreu yes, updated the converter please try again.

Dualmiracle, 2023-05-15:
Thx! it works now! now i have to look how i can get it supported for everyone.

raphaabreu, 2023-05-15:
@Koenkk, no success on my side. It still did not recognize.

I am trying to learn more about this codebase and to be able to do more complex troubleshootings. Can you point me towards any resource that would enable me to check the messages that my device is publishing (before any conversion or parsing perhaps)?

In addition to that, I don't know if this helps or not, but these are the clusters reported:

[image]

I have tried several combinations of attributes on the dev console, trying to read any data from it but it all times out:

[image]

Koenkk, 2023-05-15:
Added the device!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-05-15 by Koenkk

phoenixswiss, 2023-06-22:
Hi, I've just paired an Zigbee Model TS0222 (https://www.zigbee2mqtt.io/devices/TS0222_temperature_humidity.html) [Zigbee Manufacturer _TZ3000_kky16aay] to zigbee2mqtt (v. 1.31.2-1) but it was detected as TuYa QT-07S (https://www.zigbee2mqtt.io/devices/QT-07S.html) and all values are N/A.

[image]
[image]