Koenkk/zigbee2mqtt #17569: TS011F_plug_3 stop working after 3hours on 1.30-4

smh51, 2023-05-05:
### What happened?

After upgrade to 1.30-4 from 1.30-3, my 4 TS011F_plug_3 smart plag stop working. They froze and cannot be controlled anymore.
This happens after 3-4 hours after upgrade. Right after the upgrade, everything was working great.

### What did you expect to happen?

To be able to contol the plugs

### How to reproduce it (minimal and precise)

Upgrade from 1.30-3 to 1.30-4 and wait 3-4 hours

### Zigbee2MQTT version

1.30.4 commit: unknown

### Adapter firmware version

7.1.1.0 build 273

### Adapter

skyconnect

martinusgithubius, 2023-05-05:
I have similar problems, with TS011F (_TZ3000_gvn91tmx), z2m version 1.30.4-1. The values of 'power' and 'current' remain unchanged after a certain period, especially when the device has been turned off. The recorded values during the 'on' period are still being displayed.

smh51, 2023-05-05:
maybe it's due also to new update 2023.5.

Chronocento, 2023-05-06:
> I have similar problems, with TS011F (_TZ3000_gvn91tmx), z2m version 1.30.4-1. The values of 'power' and 'current' remain unchanged after a certain period, especially when the device has been turned off. The recorded values during the 'on' period are still being displayed.

I'm having the exact same problem since the 1.30.4-1 update, with the exact same plug (TS011F _TZ3000_gvn91tmx).

The problem occured also with HASS 2023.4, no changes since the upgrade to 2023.5.

Koenkk, 2023-05-06:
Can you check if this is fixed with the latest dev branch? (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

smh51, 2023-05-06:
For the next week i'll not be able to test it. If @Chronocento or @martinusgithubius  could test it, it would be great :)

drifter75, 2023-05-07:
> Can you check if this is fixed with the latest dev branch? (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

had the same isse with 2 of my TS011F_plug_1 (_TZ3000_gvn91tmx), where power was reporting the last value instead of 0W as the appliance is switched of. noticed this after upgrading Z2M to 1.30.4.

After upgrading to latest-dev it's back to 0W, so seems like it's fixed.

martinusgithubius, 2023-05-09:
Having installed the edge version tonight, I am now commencing its testing, and at this stage, it seems to hold great promise. Thank you, Koenkk, for your valuable support.

martinusgithubius, 2023-05-10:
after 24h  I can show a 'thumbs up'

Koenkk, 2023-05-11:
Thanks than I will close this (it is a duplicate of #16709)

closed 2023-05-11 by Koenkk