Koenkk/zigbee2mqtt #17582: [New device support]: TS011F with manufacturer name _TZ3000_okaz9tjs AKA BSD29!

atv2016, 2023-05-07:
### Link

https://www.aliexpress.com/item/1005004945906794.html?spm=a2g0o.new_account_index.0.0.64fe25b9u0Pi74

### Database entry

Friendly name 0x70b3d52b60023ba0 Description Last seen N/A Availability Disabled Device type Router Zigbee Model TS011F Zigbee Manufacturer _TZ3000_okaz9tjs Support status Not supported IEEE Address 0x70b3d52b60023ba0 Network address 0xB789 Firmware build date 20210625 Firmware version 500+TZSKT31BS102 Power Interview completed True

### Comments

Seems related to : https://github.com/Koenkk/zigbee2mqtt/issues/14284

I'm getting: Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_okaz9tjs'

On the back of the plug it says BSD29.

I have not tried adding an external converter yet as it seems it's already identified by per the GitHub link that it identifies as a TS011F Tuya Plug_1.

Would really appreciate these being added as they are 20A and i i need this for some high draw devices.

ikaruswill, 2023-05-10:
So I have the same device but it's successfully paired:
```
_TZ3000_okaz9tjs
```

Identified as:
```
TS011F_plug_3
```

atv2016, 2023-05-10:
> Hmmm interesting. Maybe I need to repair/reconfigure then

When did you set it up? Maybe it just got added?

atv2016, 2023-05-10:
What version of zigbee2mqtt are you running? Do you have a special setup or running on a dev branch?
I’m just running it big standard inside HA in a vm.

atv2016, 2023-05-10:
Just tried again no dice. This is what I see in the device info:

Zigbee Model
TS011F
Zigbee Manufacturer
_TZ3000_okaz9tjs
Support status

Not supported

IEEE Address
0x70b3d52b60023ba0
Network address
0xC106
Firmware build date
20210625
Firmware version
500+TZSKT31BS102
Maybe your firmware is newer?

atv2016, 2023-05-13:
@Koenkk Would you be able to help me get these working?

Koenkk, 2023-05-13:
All devices with model `TS011F` should be supported in the latest z2m, make sure you are running the latest z2m version (1.30.4)

atv2016, 2023-05-13:
Ah, for some reason my HA says:
Zigbee2mqtt 
Current version: 1.25.2-1 (Changelog)

Are they not including the latest version?

PastCoder, 2023-06-18:
Im am using not zigbee2mqtt but https://github.com/ioBroker/ioBroker.zigbee - which as I understood uses mainly the same code base.
My two plug _TZ3000_okaz9tjs were working for some time without any issue but stoppen 2-3 weeks ago. 
Was there any related change?

closed 2023-07-28 by github-actions[bot]