Koenkk/zigbee2mqtt #17583: [Wrong device]: Tuya ZigBee Human Presence Sensor ZY-M100-S

Burki24, 2023-05-07:
### Link

https://de.aliexpress.com/item/1005004704966668.html?aff_fcid=a6c1939fce52480b88ae7361ee630d45-1683452632112-09740-_Ddy0YiH&tt=CPS_NORMAL&aff_fsk=_Ddy0YiH&aff_platform=shareComponent-detail&sk=_Ddy0YiH&aff_trace_key=a6c1939fce52480b88ae7361ee630d45-1683452632112-09740-_Ddy0YiH&terminal_id=478e162037384eab98ba9ba82186f96e&afSmartRedirect=y&gatewayAdapt=4itemAdapt

### Model

ZY-M100-S

### Description

Tuya ZigBee Human Presence Sensor

### Vendor

Tuya

### Picture (link)

https://github.com/Burki24/Z2M_Pictures/blob/main/devices/radar-wall-1.jpg

### Database entry

{"id":47,"type":"Router","ieeeAddr":"0xe0798dfffed6de48","nwkAddr":55536,"manufId":4098,"manufName":"_TZE204_ztc6ggyl","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":".3�+i/3�+i/3�+i03�+i03�+i33�+i43�+i53�+i","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_ztc6ggyl","powerSource":1,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":"","appVersion":70}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1683450017555,"defaultSendRequestWhen":"immediate"}

### Notes

This is the wall mount Version of the ZY-M100. Ther is also an in-wall mount version with a different design called ZY-M100-L.

Koenkk, 2023-05-07:
Already fixed in https://github.com/Koenkk/zigbee2mqtt/issues/17525

closed 2023-05-07 by Koenkk

Burki24, 2023-05-07:
Sorry,

but it isn't fixed:
[image]
[image]

Running on 1.30.4-dev commit b8e46075. Tested in Edge, Chrome and firefox. With cleaned cache in every browser.

Greetings
Burkhard

ERgoliner, 2023-05-08:
the ceiling mount got detected the right way

[image]

but not the wall mount mentioned at the top

[image]

Sch8er, 2023-05-08:
> Everything works fine with this #17483, check converter: [https://gist.github.com/Koenkk/9295fc8afcc65f36027f9ab4d319ce64](url)

If i try to enter at configuration.yml
-> Integration error: external_converters - Integration 'external_converters' not found.
If i added it to Configuration of "Zigbee2MQTT Edge" no changes - still "not supported"
`data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  base_topic: zigbee2mqtt
serial:
  port: /dev/ttyUSB0
external_converters:
  - /config/ext_converter.js`

Any idea?

ERgoliner, 2023-05-08:
> Everything works fine with this #17483, check converter: [https://gist.github.com/Koenkk/9295fc8afcc65f36027f9ab4d319ce64](url)

confirmed thx all