Koenkk/zigbee2mqtt #17585: Cleverio SA100 siren mqtt message not working

kryllem, 2023-05-07:
### What happened?

Publishing warning payload as described in https://www.zigbee2mqtt.io/devices/SA100.html is not working for this device.
I can triger siren from web GUI but not by publishing a payload from HA.

I tested to publish volume payload to check if that works and it worked.
It's needed to get warning to work as the siren is useless if it can not be triggered by automation.

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

20210708

### Adapter

Sonoff CC2652P

### Debug log

Debug 2023-05-07 11:56:19Received MQTT message on 'zigbee2mqtt/0x540f57fffe85dacb/set/' with data '{"warning": {"mode": fire, "level": very_high, "strobe_level": very_high, "strobe": true, "strobe_duty_cycle": 10, "duration": 10}}'
Error 2023-05-07 11:56:19Invalid message 'null', skipping...

kryllem, 2023-05-09:
I made some more tests and got it work.
Issue was in missing "" in few places.

Example of working payload:
`{"warning":{"duration":5,"level":"very_high","mode":"fire","strobe":true,"strobe_duty_cycle":2,"strobe_level":"very_high"}}`

Recommend to update product info page.

closed 2023-05-09 by kryllem

storacm, 2024-11-17:
Hi @kryllem ,

I have the same siren but cannot get it to work. I am using node red in HA but the string you supplied does not work for me.
Do you just send the string to the mqtt topic of the SA100?