Koenkk/zigbee2mqtt #17586: [Wrong device]: Wall switch TS004 _TZ3000_mdj7kra9

jocamane, 2023-05-07:
### Link

https://www.aliexpress.com/item/1005002950510187.html

### Model

TS004 _TZ3000_mdj7kra9

### Description

 Smart light switch - 4 gang with neutral wire

### Vendor

TUYA

### Picture (link)

https://ae01.alicdn.com/kf/H7c2a226d53a64e96bfda6b1db711ae85q/Tuya-Zigbee-Smart-Light-Switch-EU-1-2-3-4-Gang-Wall-Touch-Screen-Switch-Neutral.jpg_640x640.jpg

### Database entry

{"id":65,"type":"Router","ieeeAddr":"0xa4c13823d84f7b07","nwkAddr":49227,"manufId":4417,"manufName":"_TZ3000_mdj7kra9","powerSource":"Mains (single phase)","modelId":"TS0004","epList":[1,2,3,4,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u001dI�+\u0012","65506":54,"65508":0,"65534":0,"modelId":"TS0004","stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_mdj7kra9","zclVersion":3,"appVersion":82,"powerSource":1}},"manuSpecificBosch":{"attributes":{"53251":"AAAAAgAABAAABgAA"}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff07afaa","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff07afaa","endpointID":1}],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff07afaa","endpointID":1}],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff07afaa","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":889996980},"lastSeen":1683459299211,"defaultSendRequestWhen":"immediate"}

Koenkk, 2023-05-07:
Please provide the model name listed on the device (physically), it is not `TS0004`

jocamane, 2023-05-07:
Ok.
MODEL: DS-111

Koenkk, 2023-05-08:
Thanks, fix will be included in the next release!

closed 2023-05-08 by Koenkk