Koenkk/zigbee2mqtt #17588: [Wrong device]: Smoke sensor Ts0601 _TZE200_yh7aoahi

jocamane, 2023-05-07:
### Link

At this time is unavaible

### Model

TS0601 _TZE200_yh7aoahi

### Description

Smoke Sensor

### Vendor

TUYA

### Picture (link)

https://ae01.alicdn.com/kf/S8322f166601347178d89876ffca3706d6/Temperatura-oportuna-do-salvamento-e-umidade-Sensor-voz-controle-2-4ghz-Tuya-Wifi-Smoke-Detector-2.jpg_Q90.jpg_.webp

### Database entry

{"id":60,"type":"EndDevice","ieeeAddr":"0x84fd27fffeca5ada","nwkAddr":49464,"manufId":4098,"manufName":"_TZE200_yh7aoahi","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_yh7aoahi","stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1683457528959,"defaultSendRequestWhen":"immediate"}

### Notes

The image has been changed

Koenkk, 2023-05-07:
Same as https://github.com/Koenkk/zigbee2mqtt/issues/17586#issuecomment-1537490383

jocamane, 2023-05-07:
I don't know. The box has been in the trash for a long time and the device has nothing written on it

Koenkk, 2023-05-08:
Ok, fix will be included with the next release.

closed 2023-05-08 by Koenkk

jocamane, 2023-05-08:
thanks and sorry