Koenkk/zigbee2mqtt #17590: [Wrong device]:

Sygnus83, 2023-05-07:
### Link

https://vi.aliexpress.com/item/1005003931289486.html?spm=a2g0o.order_list.order_list_main.23.3c1a1802rRQG1A&gatewayAdapt=glo2vnm

### Model

40ZH-O

### Description

Motion sensor

### Vendor

Aubess

### Picture (link)

https://ae01.alicdn.com/kf/Sa9e8e2dbe61e4581b90895fd952053d0N/AUBESS-Tuya-Zigbee-3-0-Smart-Human-Body-Sensor-Body-Movement-PIR-Motion-Sensor-Smart-Life.jpg_Q90.jpg_.webp

### Database entry

{"id":29,"type":"EndDevice","ieeeAddr":"0xa4c138536a544aeb","nwkAddr"

### Notes

I have this motion senor : https://www.zigbee2mqtt.io/devices/40ZH-O.html

but show this : https://www.zigbee2mqtt.io/devices/ZM-35H-Q.html#tuya-zm-35h-q

Koenkk, 2023-05-08:
Fixed! (will be included in the next release)

closed 2023-05-08 by Koenkk

jumavi, 2023-06-21:
I have the same problem but in reverse

I have this motion senor :  https://www.zigbee2mqtt.io/devices/ZM-35H-Q.html#tuya-zm-35h-q
but show this : https://www.zigbee2mqtt.io/devices/40ZH-O.html

jumavi, 2023-08-02:
Hi, any news about this issue? Can anyone help me?
Thanks

jantoni, 2023-09-12:
Same issue.

Occupancy is not detected