Koenkk/zigbee2mqtt #17593: [Wrong device]: MOES Smart Plug - UK  (TS011F - _TZ3000_yujkchbz)

iculookn, 2023-05-08:
### Link

https://www.aliexpress.com/item/1005004694257109.html?spm=a2g0o.order_list.order_list_main.46.4e621802uvgY2J

### Model

MOES Smart Plug

### Description

Smart Plug (with power monitoring)

### Vendor

MOES

### Picture (link)

https://ae01.alicdn.com/kf/S0d3d61222fdb4736be378f27fcac82d8f/MOES-ZigBee-Tuya-Socket-Power-Plug-16A-Smart-APP-Wireless-Socket-Outlet-Functional-Energy-Monitor-Timer.jpg_640x640.jpg

### Database entry

{"id":33,"type":"Router","ieeeAddr":"0xa4c13807a873d80e","nwkAddr":63716,"manufId":4417,"manufName":"_TZ3000_yujkchbz","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0012","65506":53,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_yujkchbz","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":1,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"powerOnBehavior":1,"switchType":0}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":235,"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsCurrent":86,"activePower":10}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,107]}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x94deb8fffeecb1a4","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x94deb8fffeecb1a4","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x94deb8fffeecb1a4","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-892627439},"lastSeen":1683512813803,"defaultSendRequestWhen":"immediate"}

### Notes

Plug does not have a model NUMBER on it, just "MOES Smart Plug"

Koenkk, 2023-05-08:
How is it currently detected, `TS011F_plug_1` or `TS011F_plug_3`?

iculookn, 2023-05-09:
Detected as TuYa TS011F_plug_1

Koenkk, 2023-05-09:
Thanks, will be fixed with the next release.

iculookn, 2023-05-11:
Thanks, I updated to 1.30.4-dev commit: [dddd2bc] and all looks good, apart from the photo.

it is linked to this, but that is a photo of a Loratap switch?
https://slsys.github.io/Gateway/devices/png/TS011F.png

Also, if I click on the model link in the device page, it goes to a blank page.

Unless it is not ready in Dev and only will be ready in the master release, then sorry for jumping the gun.

Thanks