Koenkk/zigbee2mqtt #17600: TuYa TS011F_plug_1 | Zigbee2MQTT exposing wrong values

b2Homard, 2023-05-08:
### What happened?

Hello everyone !

As i was trying to setup some simple automation using a smart plug with power measurement, and struggled a lot, I saw something I can’t understand.

I turn ON the fan plugged to the smart plug and Zigbee2MQTT is exposing 33W, which is normal.

I also check manually the measured power using “Dev Console” Tab and read 33W.

I turn OFF the fan and Zigbee2MQTT still exposes 33W, forever. It will only change after I pull some power (e.g. >20W). It will also never go down to 0W even if the fan is OFF or unplugged.

However, if I manually check the power using the “Dev Console” tab, I always have the right value(0W when the fan is OFF and ~33W when the fan is ON).
[image]
[image]
[image]

### What did you expect to happen?

Expose voltage, power and current periodically, as defined in the "Reporting" tab, no matter the load.

### How to reproduce it (minimal and precise)

Draw some power through the plug (at least 20W) then stop drawing power. Then draw less power (e.g. a small LED lamp, 6W). Observe that the exposed power (and current) is not actualizing. Manually read the consumed power to compare to the exposed power.

### Zigbee2MQTT version

 1.30.4

### Adapter firmware version

 6.10.3.0 build 297

### Adapter

Sonoff | Zigbee 3.0 | Dongle Plus-E

### Debug log

[DebugLog.txt](https://github.com/Koenkk/zigbee2mqtt/files/11421490/DebugLog.txt)

alexh3o, 2023-05-09:
Hello, I have the same problem, and according to this thread we need to wait a fix in the next z2m update
[Issue 17569](https://github.com/Koenkk/zigbee2mqtt/issues/17569)

tjorim, 2023-05-09:
Same here, Zigbee2MQTT stable version 1.30.4:

- Zigbee Model: TS011F
- Zigbee Manufacturer: _TZ3000_gvn91tmx

**Also** tested add-on **edge** version 1.30.4-dev **commit a6bebf1**, values go back to 0 but does not seem reliable.
E.g. kettle was off for hours, still took a few minutes and toggling the plug off and on a few times in z2m before it got to 0.
So seems to run better, but not as before. Will keep monitoring and try to compare to pre-1.30.4.

Koenkk, 2023-05-09:
Duplicate of #16709

closed 2023-05-09 by Koenkk

b2Homard, 2023-05-14:
I don't think that I expose the same problem in my issue ! Maybe I'm missing something though.

sved-SI, 2023-05-15:
> Duplicate of #16709

Different problem. power stuck and won't change for days!  ( even if the heater is turned off ) Switch is set to off

sved-SI, 2023-05-15:
how you check the power using the “Dev Console ?

Koenkk, 2023-05-15:
Did you already check with the latest dev branch?  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

b2Homard, 2023-05-20:
Go to Zigbee2MQTT extension. Click on a device and go in the "Dev Console" Tab. Select "haElectricalMeasurement" in "Cluster" and  select "activePower" in "Attribute". _Above the green button "Read", a line will be added._ Click the "Read" button. This value is always right (for me at least).
[image]

Koenkk, 2023-05-21:
@b2Homard the fix will be included in the next release on 1 June.

sakirsek, 2023-06-22:
> @b2Homard the fix will be included in the next release on 1 June.

hey Koenkk, the case of showing wrong current value continues. wattage is correct but current is wrong. It should be 2A but it shows 1A. 5v * 2A = 11W is correct.

[image]
[image]
[image]

network-shark, 2025-04-23:
Could it be , that the issue is back ? I do have one TS011F_plug_3 which always sends power set to 35W it was working for month. Power cycle does not help .