Koenkk/zigbee2mqtt #17604: TS0121_plug reporting wrong power values in 1.30.4

kristofejro, 2023-05-08:
### What happened?

After upgrade z2m to 1.30.4 BlitzWolf BW-SHP13 TS0121_plug reporting wrong power values.

Zigbee Model
TS0121
Zigbee Manufacturer
_TZ3000_g5xawfcq

Exactly as reported in https://github.com/Koenkk/zigbee2mqtt/issues/16709

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

20221226

### Adapter

zzh

Koenkk, 2023-05-09:
Can you provide the debug log of the plug reporting correct values vs incorrect ones?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

ashimokawa, 2023-05-11:
I had the same problem, values even got stuck on 509W. Then I noticed that I could not even switch the device on/off. Happened once a day.

Went back to 1.30.3 for now. Which seems stable.

kristofejro, 2023-05-12:
I won't be able to provide any logs now. If anyone here with that device can do that it would great. If not then I will close the issue within few days. Thanks.

ralphrtm, 2023-05-13:
Hello, I had a similar issu as @ashimokawa. All day long the smart plug seems to work perfectly (power, current, voltage, energy report correctly). But once a day the smart plug will stop report anything (I can't even turn it on or off). Restart the docker container won't fix the issu, but restart zigbee2mqtt from the web interface will bring back the smart plug. I'm not even sure if I will be able to grap debug information since when the smart plug stop reporting nothing just happen (no error notification in zigbee2mqtt) when I try to turn on or off the switch. For now, it work without a problem, I will try again when it will stop reporting back and post the report here. 

My configuration:
Host: Raspberry Pi 4
Docker Container Zigbee2MQTT version 1.30.4 commit: b2dd21e
Coordinator type EZSP v8 (ITead Sonoff Zigbee 3.0 USB Dongle Plus V2 model "ZBDongle-E")
Coordinator revision 6.10.3.0 build 297

[image]

[image]

chatainsim, 2023-05-23:
I have the same issue with a _TZ3000_rdtixbnu
[image]

kristofejro, 2023-05-23:
So maybe someone can provide debug log as Koenkk asked?

dennisschroer, 2023-05-24:
Here is some logging. It is for a Blitzwolf SHP13 with a heater attached. This is what I did:

- Enable debug logging
- Turn the plug off via de web interface of zigbee2mqtt
- Turn the plug on via de web interface of zigbee2mqtt
- Enable the heater
- Turn off the heater

It seems that the plug is sending an update with `activePower: 0`  but zigbee2mqtt is not publishing/updating the state

```
zigbee2mqtt         | Zigbee2MQTT:debug 2023-05-24 15:43:12: Received Zigbee message from 'Plug Kachel', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID null
zigbee2mqtt         | Zigbee2MQTT:debug 2023-05-24 15:43:15: Received Zigbee message from 'Plug Kachel', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID null
zigbee2mqtt         | Zigbee2MQTT:debug 2023-05-24 15:45:19: Received Zigbee message from 'Plug Kachel', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID null
zigbee2mqtt         | Zigbee2MQTT:debug 2023-05-24 15:47:12: Received MQTT message on 'zigbee2mqtt/Plug Kachel/set' with data '{"state":"OFF"}'
zigbee2mqtt         | Zigbee2MQTT:debug 2023-05-24 15:47:12: Publishing 'set' 'state' to 'Plug Kachel'
zigbee2mqtt         | Zigbee2MQTT:info  2023-05-24 15:47:12: MQTT publish: topic 'zigbee2mqtt/Plug Kachel', payload '{"child_lock":"UNLOCK","current":6.39,"energy":831.35,"indicator_mode":"off/on","linkquality":159,"power":1298,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":232}'
zigbee2mqtt         | Zigbee2MQTT:debug 2023-05-24 15:47:12: Received Zigbee message from 'Plug Kachel', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
zigbee2mqtt         | Zigbee2MQTT:info  2023-05-24 15:47:12: MQTT publish: topic 'zigbee2mqtt/Plug Kachel', payload '{"child_lock":"UNLOCK","current":6.39,"energy":831.35,"indicator_mode":"off/on","linkquality":159,"power":1298,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":232}'
zigbee2mqtt         | Zigbee2MQTT:debug 2023-05-24 15:47:13: Received Zigbee message from 'Plug Kachel', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":0}' from endpoint 1 with groupID null
zigbee2mqtt         | Zigbee2MQTT:debug 2023-05-24 15:47:15: Received MQTT message on 'zigbee2mqtt/Plug Kachel/set' with data '{"state":"ON"}'
zigbee2mqtt         | Zigbee2MQTT:debug 2023-05-24 15:47:15: Publishing 'set' 'state' to 'Plug Kachel'
zigbee2mqtt         | Zigbee2MQTT:info  2023-05-24 15:47:15: MQTT publish: topic 'zigbee2mqtt/Plug Kachel', payload '{"child_lock":"UNLOCK","current":6.39,"energy":831.35,"indicator_mode":"off/on","linkquality":183,"power":1298,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":232}'
zigbee2mqtt         | Zigbee2MQTT:debug 2023-05-24 15:47:15: Received Zigbee message from 'Plug Kachel', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
zigbee2mqtt         | Zigbee2MQTT:info  2023-05-24 15:47:15: MQTT publish: topic 'zigbee2mqtt/Plug Kachel', payload '{"child_lock":"UNLOCK","current":6.39,"energy":831.35,"indicator_mode":"off/on","linkquality":183,"power":1298,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":232}'
zigbee2mqtt         | Zigbee2MQTT:debug 2023-05-24 15:47:18: Received Zigbee message from 'Plug Kachel', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":235}' from endpoint 1 with groupID null
zigbee2mqtt         | Zigbee2MQTT:info  2023-05-24 15:47:18: MQTT publish: topic 'zigbee2mqtt/Plug Kachel', payload '{"child_lock":"UNLOCK","current":6.39,"energy":831.35,"indicator_mode":"off/on","linkquality":151,"power":1298,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":235}'
zigbee2mqtt         | Zigbee2MQTT:debug 2023-05-24 15:47:30: Received Zigbee message from 'Plug Kachel', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,83136]}' from endpoint 1 with groupID null
zigbee2mqtt         | Zigbee2MQTT:info  2023-05-24 15:47:30: MQTT publish: topic 'zigbee2mqtt/Plug Kachel', payload '{"child_lock":"UNLOCK","current":6.39,"energy":831.36,"indicator_mode":"off/on","linkquality":167,"power":1298,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":235}'
zigbee2mqtt         | Zigbee2MQTT:debug 2023-05-24 15:47:30: Received Zigbee message from 'Plug Kachel', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":3528}' from endpoint 1 with groupID null
zigbee2mqtt         | Zigbee2MQTT:info  2023-05-24 15:47:30: MQTT publish: topic 'zigbee2mqtt/Plug Kachel', payload '{"child_lock":"UNLOCK","current":3.53,"energy":831.36,"indicator_mode":"off/on","linkquality":167,"power":1298,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_versio [... cut ...] ":"idle"},"update_available":false,"voltage":235}'
zigbee2mqtt         | Zigbee2MQTT:debug 2023-05-24 15:48:05: Received Zigbee message from 'Plug Kachel', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID null
zigbee2mqtt         | Zigbee2MQTT:debug 2023-05-24 15:48:06: Received Zigbee message from 'Plug Kachel', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID null

```

dennisschroer, 2023-05-24:
This is caused by https://github.com/Koenkk/zigbee2mqtt/issues/16709. It is intended to filter out values 0, but in this case the 0 values are valid. I also never saw the issue described in the linked issue. Maybe it is some different firmware or hardware?

To fix both I think we need to add some kind of threshold: only set it to 0 after it has been 0 for some time? Or an option to disable this feature?

Koenkk, 2023-05-24:
@dennisschroer can you provide the data/database.db entry of your device?

dennisschroer, 2023-05-25:
How can I find that? Do I need to perform a query somewhere?

Koenkk, 2023-05-26:
you can find it in the z2m folder (next to `configuration.yaml`)

johndfoss, 2023-05-31:
This issue is also present with all my TS011F_plug_1

closed 2023-07-08 by github-actions[bot]

ashimokawa, 2023-07-10:
I still have that problem, with latest master, reverted back again to 1.30.3. 

One plug stuck at 293W, another one on 0.
The only thing that worked better than before was that I could at least still switch the devices on and off.

BenPortner, 2023-08-17:
Thanks @ashimokawa. I have the same problem and reverting to 1.30.3 like you suggested fixes it.

ashimokawa, 2023-10-29:
This problem persists, just tried the latest release again. Needed to go back to 1.30.3 once again.

This should affect so many users

@Koenkk 
can you reopen?

chatainsim, 2023-10-29:
Yes, Still have the same problems. And with other model.

ashimokawa, 2023-10-30:
I think there is no other way then to find out what happened between 1.30.3 and 1.30.4 in all dependencies. It could also be a change in the zigbee implementation also, since it seems all users reporting this have a "Dongle-E"

chatainsim, 2023-10-30:
I have a Conbee II. What is a "Dongle-E" ? Experimental support?

ashimokawa, 2023-10-30:
@chatainsim 
"Dongle-E" = ITead Sonoff Zigbee 3.0 USB Dongle Plus V2 (ZBDongle-E) with EFR32MG21 chip

So if you downgrade to 1.30.3 does this fix the problem for you also?

chatainsim, 2023-10-30:
It was, yes, but since I've updated to the latest version has I have not so much used of power reporting.
There was a fix, in settings (specific) to set all precision on 3 but this is not working anymore.

chatainsim, 2023-10-31:
Here an example with a TS011F_plug_3:
[image]

Koenkk, 2023-10-31:
Could you check with the following external converter: https://gist.github.com/Koenkk/3e84ac39d694364e78e7c04a439ad9aa

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m
- capture the logging (STDOUT), you should see logging like `TS011F_plug_3 SET TIMER`
- provide me the logging when it stops working.

chatainsim, 2023-10-31:
Hello @Koenkk 
I've added the converter and restarted zigbee2mqtt.
I've set the log to debug.
I will provide you the logs when I have the issue.
I have 119 zigbee devices so logs maybe a bit hudge.

chatainsim, 2023-11-01:
I don't see "TS011F_plug_3 SET TIMER" in the log file.
How can I get the STOUD ? zigbee2mqtt si manage by systemd.

Thanks

Koenkk, 2023-11-03:
Something like `sudo journalctl -u zigbee2mqtt` should do it

chatainsim, 2023-11-03:
In journalctl there is no "TS011F_plug_3 SET TIMER" either.

Koenkk, 2023-11-03:
I've updated https://gist.github.com/Koenkk/3e84ac39d694364e78e7c04a439ad9aa, it should show `TS011F_plug_3 ON EVENT` on z2m startup

chatainsim, 2023-11-03:
I still have nothing in journalctl, data/log/* or in /var/log/*

chatainsim, 2023-11-05:
Here are some logs from yesterday (04/11)
"Panneau Solaire" stop working around 16:40
Hope it's help.
[zigbee2mqtt.log](https://github.com/Koenkk/zigbee2mqtt/files/13260136/zigbee2mqtt.log)

Koenkk, 2023-11-05:
- This log file does not contain any logging like `TS011F_plug_3 SET TIMER` so it's not useful for me
- I updated https://gist.github.com/Koenkk/3e84ac39d694364e78e7c04a439ad9aa, please check the description ends with CUSTOM now (in the z2m frontend)
- Make sure that e.g `TS011F_plug_3 ON EVENT`  shows up in the logging
-

chatainsim, 2023-11-05:
Thank you.
I've changed the external converter.
Still nothing in the log (no `TS011F_plug_3 ON EVENT`) but description ends with CUSTOM so external converter is working.
[image]

I let you know when I have someting in the logs.

chatainsim, 2023-11-16:
Since 11/05, no more issue on TS011F_plug_3.
Still no `TS011F_plug_3 ON EVENT` in logs.

I still have some issue with TS0121_plug, can I adapte the external converter for this plug ?
Thanks

Koenkk, 2023-11-16:
See https://github.com/Koenkk/zigbee2mqtt/issues/19471#issuecomment-1807225061 for the `TS0121_plug`

chatainsim, 2023-11-22:
Hello,
One of my plug stopped reporting the power but still nothing in the log.
I don't know how to have logs.
I've setuped a graylog server and configured zigbee2mqtt to send log to graylog but no `TS011F_plug_3 ON EVENT` or `TS011F_plug_3 SET TIMER`.
I've also configure rsyslog to log zigbee2mqtt in a dedicated file but no ON EVENT or SET TIMER.

Koenkk, 2023-11-22:
@chatainsim how do you run z2m? The logging for this case only appears on the STDOUT

chatainsim, 2023-11-22:
I run z2m with npm, started with systemd.

Koenkk, 2023-11-22:
See [this](https://stackoverflow.com/questions/37585758/how-to-redirect-output-of-systemd-service-to-a-file) on how to configure the [z2m systemd service](https://www.zigbee2mqtt.io/guide/installation/01_linux.html#optional-running-as-a-daemon-with-systemctl) to log the STDOUT to a file

chatainsim, 2023-11-22:
I've already added this to z2m service:
```
StandardOutput=syslog
StandardError=syslog
SyslogIdentifier=Z2M
```
And in rsyslog I've redirected to a file in /var/log/z2m/z2m.log

```
root@VM-Z2M:/etc/rsyslog.d# cat z2m.conf
if $programname == 'Z2M' then /var/log/z2m/z2m.log
& stop
```

chatainsim, 2023-11-22:
I have the same output as /var/log/z2m/z2m.log.
Still no ON EVENT on startup.
I don't understand why

Koenkk, 2023-11-23:
what if you just run 'npm start' in the z2m directory instead of via systemd?

chatainsim, 2023-11-23:
It's what I've done. No ON EVENT.

Koenkk, 2023-11-23:
I messed up, sorry! Fixed https://gist.github.com/Koenkk/3e84ac39d694364e78e7c04a439ad9aa now.

chatainsim, 2023-11-23:
Hi @Koenkk , no problem.
Thanks, it's working now 👍 
```
root@VM-Z2M:/var/log/z2m# grep "TS011F_plug_3 ON EVENT" *
Nov 23 20:10:56 VM-Z2M Z2M[19330]: TS011F_plug_3 ON EVENT
Nov 23 20:10:56 VM-Z2M Z2M[19330]: TS011F_plug_3 ON EVENT
Nov 23 20:10:58 VM-Z2M Z2M[19330]: TS011F_plug_3 ON EVENT
Nov 23 20:10:58 VM-Z2M Z2M[19330]: TS011F_plug_3 ON EVENT
Nov 23 20:10:58 VM-Z2M Z2M[19330]: TS011F_plug_3 ON EVENT
Nov 23 20:11:00 VM-Z2M Z2M[19427]: TS011F_plug_3 ON EVENT
Nov 23 20:11:00 VM-Z2M Z2M[19427]: TS011F_plug_3 ON EVENT
Nov 23 20:11:00 VM-Z2M Z2M[19427]: TS011F_plug_3 ON EVENT
```
I let you know.

chatainsim, 2023-11-26:
@Koenkk here are some logs.
Tell me if it's ok. If not I will provide you all the logs.
[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/13466607/z2m.log)

Koenkk, 2023-11-26:
@chatainsim at what time does the power stop updating?

chatainsim, 2023-11-26:
@Koenkk it's hard to tell has it's for solar panels. 
[image]
Yesterday morning there was some issue with it already, restarted z2m and it stop working again between 6PM and at least 7AM on the 26th of November.

Koenkk, 2023-11-27:
I've updated the logging again, can you provide the logging again with: https://gist.github.com/Koenkk/3e84ac39d694364e78e7c04a439ad9aa

chatainsim, 2023-11-27:
Sure, no problem.

chatainsim, 2023-12-22:
Hello @Koenkk 
Here are the last log of the issue.
Plug 0x84ba20fffe6e75fc stop working around 20:10

[z2m_12212020.log](https://github.com/Koenkk/zigbee2mqtt/files/13751817/z2m_12212020.log)

Thank you.

Koenkk, 2023-12-23:
Looking at the logs, the timer keeps working, so probably the read throws an error, I've updated the logging a bit (https://gist.github.com/Koenkk/3e84ac39d694364e78e7c04a439ad9aa), please also include the regular z2m logging

chatainsim, 2023-12-23:
Here are the full log from 10 th december to today.
I've seen the logs are not in debug. Do you need debug ?
I will update the converter and provide you next full logs.
Thanks

[z2m_122120.log](https://github.com/Koenkk/zigbee2mqtt/files/13758450/z2m_122120.log)

Koenkk, 2023-12-24:
Yes with debug logging please and the next ext converter.

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

Koenkk, 2024-01-04:
Was a load attached to the plug at this time?

```
Jan  3 23:56:35 VM-Z2M Z2M[502]: TS011F_plug_3 TIMER TICK0x84ba20fffe6e75fc
Jan  3 23:56:35 VM-Z2M Z2M[502]: TS011F_plug_3 electricalMeasurement0x84ba20fffe6e75fc
Jan  3 23:56:35 VM-Z2M Z2M[502]: Zigbee2MQTT:debug 2024-01-03 23:56:35: Received Zigbee message from 'Wolf Panneau 2', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":225}' from endpoint 1 with groupID null
Jan  3 23:56:35 VM-Z2M Z2M[502]: TS011F_plug_3 ON EVENT 0x84ba20fffe6e75fc
Jan  3 23:56:35 VM-Z2M Z2M[502]: Zigbee2MQTT:info  2024-01-03 23:56:35: MQTT publish: topic 'zigbee2mqtt/Wolf Panneau 2', payload '{"child_lock":"LOCK","current":0,"energy":56.76,"indicator_mode":"off/on","last_seen":"2024-01-03T22:56:35.350Z","linkquality":255,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":225}'
Jan  3 23:56:35 VM-Z2M Z2M[502]: TS011F_plug_3 electricalMeasurement done0x84ba20fffe6e75fc
Jan  3 23:56:35 VM-Z2M Z2M[502]: TS011F_plug_3 DONE0x84ba20fffe6e75fc
```

Asking because the plug returns 0 for power.

chatainsim, 2024-01-04:
There is a solar panel attach to this plug.
But during that time value should be 0, yes 👍

Koenkk, 2024-01-04:
when did you attach a load again?

chatainsim, 2024-01-04:
Solar panel is always attach.
But it should have started the load on the 4th around 9AM.

Koenkk, 2024-01-04:
It looks like z2m does this indeed, e.g.:

```
Jan  4 12:35:55 VM-Z2M Z2M[502]: TS011F_plug_3 TIMER TICK0x84ba20fffe6e75fc
Jan  4 12:35:55 VM-Z2M Z2M[502]: TS011F_plug_3 electricalMeasurement0x84ba20fffe6e75fc
Jan  4 12:35:55 VM-Z2M Z2M[502]: Zigbee2MQTT:debug 2024-01-04 12:35:55: Received Zigbee message from 'Wolf Panneau 2', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":274,"rmsCurrent":1170,"rmsVoltage":230}' from endpoint 1 with groupID null
Jan  4 12:35:55 VM-Z2M Z2M[502]: TS011F_plug_3 ON EVENT 0x84ba20fffe6e75fc
Jan  4 12:35:55 VM-Z2M Z2M[502]: Zigbee2MQTT:info  2024-01-04 12:35:55: MQTT publish: topic 'zigbee2mqtt/Wolf Panneau 2', payload '{"child_lock":"LOCK","current":1.17,"energy":57.44,"indicator_mode":"off/on","last_seen":"2024-01-04T11:35:55.878Z","linkquality":252,"power":274,"power_outage_memory":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":230}'
Jan  4 12:35:55 VM-Z2M Z2M[502]: TS011F_plug_3 electricalMeasurement done0x84ba20fffe6e75fc
Jan  4 12:35:55 VM-Z2M Z2M[502]: TS011F_plug_3 DONE0x84ba20fffe6e75fc
Jan  4 12:35:55 VM-Z2M Z2M[502]: TS011F_plug_3 SET TIMER0x84ba20fffe6e75fc
```

274W is published at Jan  4 12:35:55

chatainsim, 2024-01-04:
Yes, this one was working, I forget to mention that.
But not the other one : 0x84ba20fffe6e7c71

Koenkk, 2024-01-04:
I see the issue indeed, could you provide the **herdsman debug** log of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Also, what was the last known working version?

chatainsim, 2024-01-04:
Thanks @Koenkk 
I've enabled the debug logs for herdsman.
I'm sorry, I can't remenber the last working version. Maybe around 1.30, not sure.

[2 more replies]