Koenkk/zigbee2mqtt #17606: Wrong temperature range in LED controller - Skydance WZ1

chojnicki, 2023-05-08:
### What happened?

I bought Skydance WZ1, a CCT led strip controller on aliexpress. My light strips are 2700-6500K. On aliexpress seller page or manual range is not specified, but I found one shop with that item where it claims range exactly from 2700K. Just to be sure I wrote email directly to manufacturer Skydance, and surprisingly got answer. I did not mentioned 2700K, I just asked about temperature range and they replied it's 2700-6500K.

So I got my controller, connected it and noticed that range in Z2M is to 500 mired, to 2000K like in most controllers, not 2700K. It is detected as [TS0502B](https://www.zigbee2mqtt.io/devices/TS0502B.html#tuya-ts0502b) _TZB210_nfzrlz29. After setting light in HA to for example 3000K, I'm getting much colder light. So get real 3000K I have to set 400 mired, not 333 mired like it should be.

I checked [converter](https://github.com/Koenkk/zigbee-herdsman-converters/blob/15ba61d36751c3a687989c9bff9388d37d1f8f84/devices/tuya.js#L2010), and there is hardcoded temperature range:

`extend: tuya.extend.light_onoff_brightness_colortemp({colorTempRange: [153, 500], noConfigure: true}),`

So I tried to make custom converter with different colorTempRange. Even that Z2M dashboard had correct slider, it did calibrated anything. I found in sources that for some devices it actually tries to detect supported range by reading clusters. First time I was using that feature, but i managed to execute in dev console `LColorCtrl` => `colorTempPhysicalMax` and got result.

`Info 2023-05-08 21:48:45Read result of 'lightingColorCtrl': {"colorTempPhysicalMax":500}`

 I'm confused. Please explain to me - does it comes directly from device? It supposed to be 370 mired, so it makes no sense. Anything could be done from Z2M side to calibrate this? I tried to write that attribute, but got error that it's read only. 

### What did you expect to happen?

Different color range

### How to reproduce it (minimal and precise)

Pair device.

### Zigbee2MQTT version

1.30.3-1

### Adapter firmware version

26720700

### Adapter

ConBee2

Koenkk, 2023-05-09:
> I'm confused. Please explain to me - does it comes directly from device? It supposed to be 370 mired, so it makes no sense. Anything could be done from Z2M side to calibrate this?

this value indeed comes from the device, it is programmed in its firmware.

closed 2023-06-17 by github-actions[bot]

fatshotty, 2024-10-08:
hi, i have a wz1 set in CCT mode, but z2m works in singol color mode. How can i enable the cct mode in z2m?
I tried to add `LColorCtrl -> colorTemperature` but it gives me an error :/

thanks in advance