Koenkk/zigbee2mqtt #17608: [New device support]: Zigbee Solar Powered Siren - _TZE200_nlrfgpny

cloudbr34k84, 2023-05-08:
### Link

https://www.aliexpress.com/item/1005005264809268.html?spm=a2g0o.detail.0.0.58192fb4f65Tv4&gps-id=pcDetailBottomMoreThisSeller&scm=1007.13339.291025.0&scm_id=1007.13339.291025.0&scm-url=1007.13339.291025.0&pvid=df380b42-20e8-48e7-9cef-fecb8501883a&_t=gps-id:pcDetailBottomMoreThisSeller,scm-url:1007.13339.291025.0,pvid:df380b42-20e8-48e7-9cef-fecb8501883a,tpp_buckets:668%232846%238114%231999&isseo=y&pdp_npi=3%40dis%21AUD%2171.51%2146.47%21%21%21%21%21%402101c71316835891663986923ef005%2112000032412568207%21rec%21AU%21762881936

### Database entry

{"data":{"friendly_name":"0xa4c1383a9082f6d1",'NWK 0x623C ""ieee_address":"0xa4c1383a9082f6d1","status":"started"},"type":"device_interview"}

### Comments

I did not do this,

ParalaX002, 2023-05-09:
Hi,

There is already some work in progress here:
https://github.com/Koenkk/zigbee2mqtt/discussions/17325#discussioncomment-5829438

It is working, but we're missing the battery level.

cloudbr34k84, 2023-05-09:
> Hi,
> 
> There is already some work in progress here:
> https://github.com/Koenkk/zigbee2mqtt/discussions/17325#discussioncomment-5829438
> 
> It is working, but we're missing the battery level.

Man I did a search and nothing came up,  so thanks!  👌

closed 2023-05-13 by cloudbr34k84

Gics78, 2023-08-06:
Tamper switch not supported, so can't be installed physically without triggering the alarm (need to deactivate tamper switch first and then reactivate).
Can anybody help? I have the device but unfortunately don't own a tuya gateway.