Koenkk/zigbee2mqtt #17610: [Wrong device]: Incorrectr Picture

cloudbr34k84, 2023-05-09:
### Link

https://www.zemismart.com/products/zm85el-2z

### Model

_TZE200_cf1sl3tj

### Description

Zemismart Tuya Zigbee Electric Curtain Robot Rechargeable Motor For Roman Rod I Type U Type Tracks Alexa Google Home Control

### Vendor

Tuya

### Picture (link)

https://ueeshop.ly200-cdn.com/u_file/UPAC/UPAC387/2210/photo/1b73914871.jpg?x-oss-process=image/format,webp

### Database entry

EndDevice, TS0601, _TZE200_cf1sl3tj, 0x9035eafffe03d467,  0x15BA

Koenkk, 2023-05-09:
please provide the model written on the device, not `_TZE200_cf1sl3tj` or `TS0601`

cloudbr34k84, 2023-05-09:
> please provide the model written on the device, not `_TZE200_cf1sl3tj` or `TS0601`

Gotcha,  done

Koenkk, 2023-05-10:
Thanks, will be fixed with the next release.

closed 2023-05-10 by Koenkk