Koenkk/zigbee2mqtt #17612: Wrong Picture

cloudbr34k84, 2023-05-09:
### Link

https://www.zigbee2mqtt.io/devices/TS0505B_1.html

### Model

TS0505B

### Description

Zigbee 3.0 18W Led Light Bulb E27 RGBCW

### Vendor

Tuya

### Picture (link)

https://images.app.goo.gl/F4qnmT7gxPxzBFT76

### Database entry

_TZ3210_jd3z4yig ; 0xa4c13849351af9e9:;  0xB4D1

Koenkk, 2023-05-09:
Please provide the model number written on the device or advertised by the seller.

cloudbr34k84, 2023-05-09:
> Please provide the model number written on the device or advertised by the seller.

There is no model number on the device except Zigbee E27 Smart Bulb. 

Should we use the official model number supplied by tuya considering there are so many vendors that also their own name on these bulbs and resell.

Koenkk, 2023-05-10:
I will use `TS0505B_1_1` in that case, thanks!

closed 2023-05-10 by Koenkk

cloudbr34k84, 2023-05-13:
Hey @Koenkk 
Just updated and noticed some bulbs didn't update 

Below is an updated one

[image]

Here is a bulb that didn't update. It has slightly different zigbee Manufacturer and model number.  It's the same bulb?? 
[image]

lucasmarcondes, 2023-11-03:
@Koenkk I have the same issue. I have 4 TS0505B bulbs: 3 use model TS0505B_1_1 and 1 uses model TS0505B_1

I believe they all should use TS0505B_1_1. The manufacturer that's off is the same as above, `_TZ3210_r5afgmkl`.

Koenkk, 2023-11-03:
@lucasmarcondes pushed a fix, will be included in the next release.

lucasmarcondes, 2023-11-10:
Hey @Koenkk, so sorry to bother but the `_TZ3210_r5afgmkl` device is still showing as TS0505B_1 instead of TS0505B_1_1. Do you think the model needs to be included in the fingerprint field as well?

Koenkk, 2023-11-10:
Made a mistake, pushed another fix.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)