Koenkk/zigbee2mqtt #17614: Template variable warning: 'dict object' has no attribute 'preset'

Bastian007, 2023-05-09:
### What happened?

After Restart of Home Assistant I receive warning messages as below.

`Logger: homeassistant.helpers.template
Source: helpers/template.py:716
First occurred: 23:21:11 (32 occurrences)
Last logged: 23:35:18
Template variable warning: 'dict object' has no attribute 'preset' when rendering '{{ value_json.preset }}'`

I believe this message belongs to Moes BHT series Thermostat (BHT-002-GCLZB) as I can find in the zigbee2mqtt log the following message:
`info  2023-05-05 00:23:58: MQTT publish: topic 'homeassistant/climate/0x04cd15fffe26efa3/climate/config', payload '{"action_template":"{% set values = {None:None,'idle':'off','heat':'heating','cool':'cooling','fan_only':'fan'} %}{{ values[value_json.running_state] }}","action_topic":"zigbee2mqtt/Moes Thermostat Dachstudio","availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Moes Thermostat Dachstudio/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","current_temperature_template":"{{ value_json.local_temperature }}","current_temperature_topic":"zigbee2mqtt/Moes Thermostat Dachstudio","device":{"identifiers":["zigbee2mqtt_0x04cd15fffe26efa3"],"manufacturer":"Moes","model":"Moes BHT series Thermostat (BHT-002-GCLZB)","name":"Moes Thermostat Dachstudio"},"json_attributes_topic":"zigbee2mqtt/Moes Thermostat Dachstudio","max_temp":"35","min_temp":"5","mode_command_topic":"zigbee2mqtt/Moes Thermostat Dachstudio/set/system_mode","mode_state_template":"{{ value_json.system_mode }}","mode_state_topic":"zigbee2mqtt/Moes Thermostat Dachstudio","modes":["off","heat"],"name":"Moes Thermostat Dachstudio","preset_mode_command_topic":"zigbee2mqtt/Moes Thermostat Dachstudio/set/preset","preset_mode_state_topic":"zigbee2mqtt/Moes Thermostat Dachstudio","preset_mode_value_template":"{{ **value_json.preset** }}","preset_modes":["hold","program"],"temp_step":1,"temperature_command_topic":"zigbee2mqtt/Moes Thermostat Dachstudio/set/current_heating_setpoint","temperature_state_template":"{{ value_json.current_heating_setpoint }}","temperature_state_topic":"zigbee2mqtt/Moes Thermostat Dachstudio","temperature_unit":"C","unique_id":"0x04cd15fffe26efa3_climate_zigbee2mqtt"}'`

Any advise to reslolve the issue is welcome!

### What did you expect to happen?

No warnings

### Zigbee2MQTT version

1.30.4-1

### Adapter firmware version

Z-Stack_3.x.0 coordinator 20221226

### Adapter

Sonoff 3.0 CC2652P

Bastian007, 2023-06-11:
Still present also with HA 2023.6 and Zigbee2MQTT 1.31.2-1.

Bastian007, 2023-07-12:
Issue still present with HA 2023.07.1 and Zigbee2MQTT 1.32.1

closed 2023-08-21 by github-actions[bot]

robvanoostenrijk, 2024-04-02:
This can often be resolved by changing the preset setting of the device on the z2m interface on `Exposes` tab.
Afterwards make sure the JSON on the `State` tab has a value specified for `preset`.

Other solution would be to provide a default value for `present` in the configure element perhaps.