Koenkk/zigbee2mqtt #17621: Cover AM43-0.45/40-ES-EB [_TZE200_zah67ekd] do not report state after HA restart

reaper7, 2023-05-09:
### What happened?

After upgrade z2m to 1.30.4 and after rebooting HA all my covers (working on AM43-0.45/40-ES-EB [_TZE200_zah67ekd]) do not report their state correctly. All show status unknown. I have to manually trigger the movement for the correct state to appear.

[image]

[image]

[image]

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

20221226

### Adapter

Electrolama

cloudbr34k84, 2023-05-10:
I noticed this as well. Didn't think anything of it.

closed 2023-06-17 by github-actions[bot]

efranulic, 2025-02-06:
I have the dame issue