Koenkk/zigbee2mqtt #17623: Irrigation Valve Model ZVG1 manufactured by RTX  error when closing.

dave21w, 2023-05-10:
### What happened?

I have 2 of these valves installed.
https://zigbee.blakadder.com/RTX_ZVG1.html
The valve does operate correctly when an open command is sent but when it is told to close it does shut off but Z2M always shows up this error on screen and in the log.

warning 2023-05-10 07:35:25zigbee-herdsman-converters:RTXZVG1Valve: NOT RECOGNIZED DP #6 with data {"dp":6,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}

Both valves do the same thing so I guess its a software issue, can anyone tell (I'm no programmer) from lookng at the error if this is a major issue, I bought these as I go on holiday soon so I need them to be reliable.

### What did you expect to happen?

On /Off with no errors.

### How to reproduce it (minimal and precise)

Just pair it and it will show up when you close the valve.

### Zigbee2MQTT version

1.30.4 commit: cdf62ea5

### Adapter firmware version

20211217

### Adapter

Electrolama zig-a zig-ah zStack3x0

### Debug log

warning 2023-05-10 07:35:25zigbee-herdsman-converters:RTXZVG1Valve: NOT RECOGNIZED DP #6 with data {"dp":6,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}

dave21w, 2023-06-01:
OK thanks, not sure how you are trying to use the timers. I'm using Domoticz as my automation controller and the valves work fine with this probably because I'm using the Domtocz timers to send the on / off commands. The timers I see in Z2M seem to be internal to the unit and so there status is just being shown. It seems they are there as a backup as they are countdown timers so in the event the valve does not get a close signal it will shut after the default 10 mins anyway.
Been running mine now for a few weeks without issues other than the error being reported on close, battery life looks good so far still at 100%
Instead of setting an On and Off time I found it easier to set an automatic off of 1 min therefore I only need to set an On time, currently only running twice a day unless the weather warms up some more.
Off topic I know but you may be interested, I bought some cheap inline water filters from amazon as you can undo them and fill them with a soluble plant food so everthing gets a feed as well for the first couple of waterings, I do this once a week and it works great, so much quicker then spending an hour with a watering can and plant food !!

Gura74, 2023-06-03:
I also use On/Off as workaround realizing the timers in fhem. However, I was referring to the device internal normal and cycle timers. I set them in the Z2M frontend, e.g. 
[image]
In the log I see the following
[image]
After that the device is “blocked”, i.e. I cannot switch it on and off again – also not manually. A minute or later I see the following in the log: 
[image]

Removing the batteries makes the device operable again. I plan to set up a zigbee sniffer and ordered a tuya compatible zigbee gateway for analysis. The revision of my device is labeled as “Rev1_2021_11_15_TS-EB” and identifies with _TZE200_2wg5qrjy. What is the exact model you are using? Thanks for the tip with the inline water filter.

dave21w, 2023-07-04:
Never had a problem with mine not responding / being unuseable, good job really as I'm only just back from vacation so would have had a lot of dead plants other wise. I have never used or changed any of the z2m values regarding timers etc, only send on / off commands to it and use the timers in my Domoticz install. This is what I see on the about page in Z2M for the device

Zigbee Model              TS0601
Zigbee Manufacturer  _TZE200_81isopgh
Description                 Zigbee smart water valve
Support status            Supported
Manufacturer             RTX
Model                        ZVG1

Not too sure where you are getting the revision label from, if thats on the box / packaging mine has long gone. Battery seems to drop to 75% fairly quickly but need to wait and see if that continues down at the same rate. I only had mine for 2 or 3  weeks before my holiday and as they had dropped to 75% by the time I was due to go away I changed them again to make sure they would last so it's been about 3 weeks since they were last changed, have to see how they go. The one thing I have noticed is I get all the settings / values I need in Domoticz but no battery state is reported, I have to look at Z2M for that.

closed 2023-08-13 by github-actions[bot]