Koenkk/zigbee2mqtt #17627: Smart plug with power monitoring TS011F_plug_1 reports non-zero power/current when there is no consumption

ra-kal, 2023-05-10:
### What happened?

The smart plug with power consumption measurement TS011F suddenly starts reporting non-zero values when there is real consumption. It looks like it just reporst the last non-zero value of power and current, before it goes to 0. I think this started with some Z2M update, or I'm not aware about this behavior earlier. I have 4 this plugs and all behaves same.

I have not yet tested dev branch.

{
    "child_lock": "UNLOCK",
    **"current": 0.08,**
    "energy": 7.41,
    "indicator_mode": "off/on",
    "last_seen": 1683710149435,
    "linkquality": 144,
    **"power": 5,**
    "power_outage_memory": "restore",
    "state": "ON",
    "update": {
        "installed_version": 192,
        "latest_version": 192,
        "state": "idle"
    },
    "voltage": 232,
    "elapsed": 154002,
    "update_available": null
}

### What did you expect to happen?

When there is no consumption, the power and current values should be 0

### How to reproduce it (minimal and precise)

Conect some device to the plug, that generates some consumption. Disconnect the device or switch off the plug output. The plug keeps reporting the last non-zero power and current values.

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-E

Ultrawipf, 2023-05-10:
Same issue with the same version started here a few days ago as well and it gets worse with time.
First it worked fine, then it reported 4W when idle, now always 6W (0.1A) with no device connected.

I would like to know if it is possible to do a full factory reset with calibration procedure or if there is a firmware update in case this is a common fixable issue.

In my case it may also be component degradation instead of a software issue as it was used outdoors for a few weeks but i see no way to open up the device without destroying the shell for inspection.

<details>
<summary>State report</summary>

```
{
    "child_lock": "UNLOCK",
    "current": 0.1,
    "energy": 1,
    "indicator_mode": "off/on",
    "linkquality": 127,
    "power": 6,
    "power_factor": null,
    "power_outage_memory": "off",
    "power_phase_c": null,
    "state": "OFF",
    "update": {
        "installed_version": 192,
        "latest_version": 192,
        "state": "idle"
    },
    "update_available": null,
    "voltage": 227,
    "action": ""
}
```
</details>

Koenkk, 2023-05-10:
Duplicate of #16709

closed 2023-05-10 by Koenkk

Koenkk, 2023-05-10:
Can you check with the latest dev?  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

ra-kal, 2023-05-10:
Ok, now I see the problem is caused by the modifications implemented for the "duplicate" issue.

It took some time, but at the end the value is 0.

Anyway, it would be best, if the user has the possibility to enable/disable the zero value filtering. I'm not aware about any dropouts in my case, but I was not actively searching for them. But I do not like the delay caused by capturing several values, each taking about 2mins, to the change the value to 0.

fherreror, 2024-03-05:
I have the same problem. Suddenly it reports 0 value for at least 1 minute. I have an automation to turn the switch off if it repports 0 for 1 minute and suddenly my computer turns off...