Koenkk/zigbee2mqtt #17632: BMCT-SLZ Bosch Bosch Light/shutter control unit II  still not supported?

walter-ve, 2023-05-10:
### What happened?

Installed version 1.30.4 which reports BMCT-SLZ Bosch Bosch Light/shutter control unit II  is supported.  

I paired this device.  Z2M still reports the device as unsupported.  

### What did you expect to happen?

According to https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.30.4 the BMCT-SLZ Bosch Bosch Light/shutter control unit II (@slugzero @smarts4home) is not supporter

### How to reproduce it (minimal and precise)

remove the device from device list; 're-pair' and the device comes back as unsupported.  

### Zigbee2MQTT version

1.30.4 

### Adapter firmware version

 6.10.3.0 build 297

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Debug log

Debug 2023-05-10 21:24:38Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"n9uf4-2","value":true}'
Info 2023-05-10 21:24:38Zigbee: allowing new devices to join.
Debug 2023-05-10 21:24:38Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,20,2,11,254,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
Info 2023-05-10 21:24:38MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"n9uf4-2"}'
Info 2023-05-10 21:24:39Device '0xf4b3b1fffe110267' joined
Info 2023-05-10 21:24:39MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xf4b3b1fffe110267","ieee_address":"0xf4b3b1fffe110267"},"type":"device_joined"}'
Info 2023-05-10 21:24:39Starting interview of '0xf4b3b1fffe110267'
Info 2023-05-10 21:24:39MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xf4b3b1fffe110267","ieee_address":"0xf4b3b1fffe110267","status":"started"},"type":"device_interview"}'
Debug 2023-05-10 21:24:41Received Zigbee message from 'Woonkamer Zuid Zonneblind', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
Debug 2023-05-10 21:25:02Received Zigbee message from 'Sonoff Thermometer 1', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":990}' from endpoint 1 with groupID 0

hubkraft, 2023-05-13:
It's supported, at least the light switch component which I'm using.
Did you added the device via *install code* in: zigbee2mqtt frontend> Settings> Tools> Add install code. As described in the doc of the device under [Pairing](https://www.zigbee2mqtt.io/devices/BMCT-SLZ.html#pairing)

walter-ve, 2023-05-13:
Thanks for the feedback!
I indeed missed that comment about the 'install code'. 
Note; maybe we should add somewhere an indication there is specific action needed to get the device working.  So that easy-go'ers like me note they really have to do something ;-).

walter-ve, 2023-05-13:
I added the install code as [hubkraft](https://github.com/hubkraft) indicated and now the device is recognized.

closed 2023-05-13 by walter-ve

bigbeka, 2024-12-21:
@hubkraft and @walter-ve could you please share your experience using BMCT-SLZ via Z2M? Once you paired the relay, does it turn on/off the light from home assistant AND using the S1/S2? Does BMCT-SLZ support button instead of regular wall switches? 

Documentation of BMCT-SLZ says that S1/S2 will not work if BMCT-SLZ is not paired using the Bosch Smart Home Controller ([page 12](https://www.bosch-smarthome.com/res/web/produkt/licht_rolladensteuerung2/pdf/8-750-002-899_Light_Shutter_OM_11L_V001_20240604.pdf))

[image]

In contrast:
I invested into multiple BMCT-RZ (Bosch Smart Relay) and the relays don’t respond to S1 after pairing with Z2M (with install code). They work as expected from within the home assistant. This issue is reported and open here (#23088).

Any help would be greatly appreciated!

walter-ve, 2024-12-21:
@bigbeka 
Maybe better to reacto to an existing topic in the Home Assistant forum for example:
https://community.home-assistant.io/t/bosch-shutter-control-2/453415

My experience is that these devices works perfectly.  But i use them first of all to control roller blinds. I have them not in place as switch. I have tested the switch function in the past.

bigbeka, 2024-12-21:
@walter-ve thank you for your reply and for the link. I will look into it further.