Koenkk/zigbee2mqtt #17633: [Wrong picture]: TS0202_1 vs 809WZT

posreg, 2023-05-10:
### Link

https://pl.aliexpress.com/item/1005005338281792.html?pdp_npi=2%40dis%21PLN%21138%2C64z%C5%82%2195%2C66z%C5%82%21%21%21%21%21%40211b813c16837459879373110e339d%2112000032663003650%21btf&_t=pvid%3Ae7ffe8af-cf93-4d89-af91-33713d8d0f36&afTraceInfo=1005005338281792__pc__pcBridgePPC__xxxxxx__1683745988&spm=a2g0o.ppclist.product.mainProduct&gatewayAdapt=glo2pol

### Model

809WZT

### Description

Motion Sensor

### Vendor

Tuya

### Picture (link)

https://ibb.co/9Yvybsj

### Database entry

{"id":13,"type":"EndDevice","ieeeAddr":"0xa4c1380f478120e0","nwkAddr":45992,"manufId":4417,"manufName":"_TZ3040_bb6xaihh","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,3,4,1280,0],"outClusterList":[4,6,4096,25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0202","manufacturerName":"_TZ3040_bb6xaihh","powerSource":3,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0014da4b33","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0014da4b33","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":491743244},"lastSeen":1683695415537,"defaultSendRequestWhen":"immediate"}

### Notes

Model name taken from the box

[image]

Koenkk, 2023-05-11:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-05-11 by Koenkk

kzajac83, 2023-05-12:
Hello

Just now so 17h after last post updated edge (uninstall, click check for updates and install egde again and launch it).

Unfourtunetly picture still old.

Koenkk, 2023-05-12:
@kzajac83 does it show the correct model in the frontend? (809WZT)

posreg, 2023-05-12:
> @kzajac83 does it show the correct model in the frontend? (809WZT)

it is still visible as TS0202_1
[image]

Koenkk, 2023-05-12:
Made a mistake, fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

kzajac83, 2023-05-12:
Thank you, now picture is correct on Edge and model too but link jump to not exit page / 404 error

https://www.zigbee2mqtt.io/devices/809WZT.html#tuya-809wzt

Koenkk, 2023-05-13:
@kzajac83 device page will be generated after the next release (page does not exist since it is not supported in the latest release yet)

kzajac83, 2023-06-02:
After yesterday release the web page appear however there is strange think, on edge all is working fine but on prod version 1.31.0-1 from yesterday there is "Support status: Not supported"

[image]

Anashost, 2023-06-05:
@kzajac83 already fixed, until next release you can use this external converter which works perfectly: https://github.com/Koenkk/zigbee2mqtt/issues/17364#issuecomment-1566715618

juandelarocha, 2024-01-10:
I have just bought a 809WZT and won't join my zigbee network. Zigbee2MQTT version is 1.35.1-1. Any advice?

22nra, 2024-01-15:
> I have just bought a 809WZT and won't join my zigbee network. Zigbee2MQTT version is 1.35.1-1. Any advice?

Yes I have same a similar problem, I did manage to get my TuYa 809WZT to pair fine at first. I was testing ver 1.35.1 & force removed the sensor but since then it will no longer pair any more. I tried clearing Z2M and it still wont pair, other Tuya sensors (temp/hum and door sensor) pair fine.

Anashost, 2024-01-15:
Mine still works fine!

[image]

Does your's have the same info:
* Zigbee Model
TS0202
* Zigbee Manufacturer
_TZ3040_bb6xaihh

22nra, 2024-01-15:
Hi @Anashost and @juandelarocha . I got my TuYa 809WZT to pair again today and it's back working fine. I had to move the sensor very close to the zigbee 2531 to get pairing to work. I might have an issue with my sensor as the linkquality was showing 1 then it jumped upto 92. Using latest ver1.35.1