Koenkk/zigbee2mqtt #17634: No Network Route with good signal on SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

sshaikh, 2023-05-10:
### What happened?

Two aqara switches keep falling foul at random intervals of "no network" errors independently and at different times. They do however still appear on the network map, with a fluctuating lqi.

A re-pair fixes them back to full functionality.

### What did you expect to happen?

Before my most recent z2m update (including stick fw) everything was working wonderfully and stable.

### How to reproduce it (minimal and precise)

Each switch will independently last a day or two before dropping off. On a good week they'll last for a good while.

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P  CC2652P

### Debug log

To be provided, but it's the "no network route" error.

mbak77, 2023-05-15:
I have very similar issue in a last few days. I'm using CC2652P with z2m version 1.30.4-1 and lately 2 of my Aqara RSD-M01 are loosing connection with "No network route (205)" error. I've never seen it before, but now it's happening almost everyday, although they have pretty decent LQI to coordinator. Another thing is I can't simply reset them - after reset Z2M's pop-up about "joining device" is displayed almost instantly, but it keeps trying to connect, unsuccesfully. I have to take it off the mount, bring it closer to the coordinator, then it's connecting and after that working fine for a while.
My mesh network is pretty decent, so even if it couldn't connect directly to coordinator, there's definitely a good route through other routers.

sshaikh, 2023-05-15:
I've reverted to the last known firmware that gave me some stability. If I still see the error then I suppose it's a regression in z2m itself.

sshaikh, 2023-05-19:
Well these things are always difficult to claim conclusively but I've not had any issues after 5 days or so of going back to 20210708.

@mbak77 would be interesting to see if you find the same, considering your issues seem a bit more apparent than mine.

mbak77, 2023-05-19:
I've stayed with current FW, didn't have much time to fight with this issue at the moment. It was happening to ~2 out of 6 rollers each day for a few days, however it stabilized lately and since Wednesday I didn't face any connection issues. Only thing I did was moving coordinator few centimeters to the other side...

sshaikh, 2023-05-19:
Yeah, I definitely think there's an environmental aspect to this that's now being emphasised somehow.

mbak77, 2023-05-19:
Yeah, I've had some zigbee network issues before, so it might be environmental problem as well. I checked connection map and currently those rollers finally are connected through the router that should be the most obvious option from the beginnig, however earlier they were often connected directly to the coordinator with poor lqi. Of course repairing through specific router didn't work in a long term, so I couldn't force this route anyhow.

D-moe, 2023-05-22:
Having similar issues after updating to new firmware as well. Going to attempt reverting back to improve stability

ecto1a2003, 2023-05-31:
> Well these things are always difficult to claim conclusively but I've not had any issues after 5 days or so of going back to 20210708.
> 
> @mbak77 would be interesting to see if you find the same, considering your issues seem a bit more apparent than mine.

Could you link me to the specific firmware? I'm having the same issues.

sshaikh, 2023-05-31:
I had a look and can't seem to find an active link - I had it lying around from a previous flash.

Here's the ones I found:
https://github.com/Koenkk/Z-Stack-firmware/releases

Change log:
https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/CHANGELOG.md

sshaikh, 2023-05-31:
Perhaps @Koenkk can assist

closed 2023-07-08 by github-actions[bot]