Koenkk/zigbee2mqtt #17636: Setting temperature_precision and humidity_precision has no effect for IKEA E2112 Vindstyrka

shrx, 2023-05-10:
### What happened?

I connected IKEA E2112 Vindstyrka to my zigebee gateway and I'm unable to set precision for temperature and humidity reports. By default the temperature_precision and humidity_precision parameters were set to "2" but the values are reported as precision "0" (integers). Changing the precision to "1" has no effect.

Example report:
```
May 10 22:30:11 raspberrypi npm[1609686]: Zigbee2MQTT:info  2023-05-10 22:30:11: MQTT publish: topic 'zigbee2mqtt/windstrike', payload '{"humidity":59,"linkquality":73,"pm25":7,"temperature":24,"voc_index":84}'
```

### What did you expect to happen?

I expect to be able to set the precision for the temperature and humidity readouts, as documented here https://www.zigbee2mqtt.io/devices/E2112.html#options

### How to reproduce it (minimal and precise)

Connect and start listening to IKEA E2112 Vindstyrka reports

The reports for temperature and humidity have 0 decimal precision (are integers)

### Zigbee2MQTT version

1.30.4-dev commit: 307e0dca

### Adapter firmware version

20211115

### Adapter

CC2531

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/11445770/log.txt)

crowbarz, 2023-05-11:
According to the [hardware spec for the sensor](https://sensirion.com/products/catalog/SEN54) (from https://github.com/Koenkk/zigbee2mqtt/issues/16717#issuecomment-1464989845 ) the underlying temperature hardware doesn't support greater than 0.45°C precision, so you're never going to get much more than integer precision anyway. It also says that the humidity sensor is accurate only to ~4.5%.

You may be able to check the actual values received from the sensor itself for yourself by setting the logging level to debug, wait for the next Zigbee update from the sensor, and see what raw data is being sent by the sensor.

shrx, 2023-05-11:
You're confusing accuracy with precision. Both temperature and humidity readouts can be calibrated, so that's not issue.

shrx, 2023-06-11:
Issue is still present

nicopasla, 2023-06-16:
I have the same issue, it doesn’t matter how much you put in the setting

tdse13, 2023-07-20:
same here with firmware 1.0.11 20230202

Koenkk, 2023-07-24:
Can someone provide the debug log ones the device reports humidity/temperature?

tdse13, 2023-07-24:
`Zigbee2MQTT:info  2023-07-24 19:49:05: MQTT publish: topic 'zigbee2mqtt/CH Wohnen', payload '{"device":{"applicationVersion":16,"dateCode":"20230202","friendlyName":"CH Wohnen","hardwareVersion":1,"ieeeAddr":"0xbc026efffe8b320c","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E2112","networkAddress":30392,"powerSource":"Mains (single phase)","softwareBuildID":"1.0.11","stackVersion":106,"type":"Router","zclVersion":8},"humidity":53,"last_seen":"2023-07-24T19:49:05+02:00","linkquality":200,"pm25":6,"temperature":24,"update":{"installed_version":16777233,"latest_version":16777233,"state":"idle"},"update_available":null,"voc_index":344}'
Zigbee2MQTT:info  2023-07-24 19:50:05: MQTT publish: topic 'zigbee2mqtt/CH Wohnen', payload '{"device":{"applicationVersion":16,"dateCode":"20230202","friendlyName":"CH Wohnen","hardwareVersion":1,"ieeeAddr":"0xbc026efffe8b320c","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E2112","networkAddress":30392,"powerSource":"Mains (single phase)","softwareBuildID":"1.0.11","stackVersion":106,"type":"Router","zclVersion":8},"humidity":53,"last_seen":"2023-07-24T19:50:05+02:00","linkquality":200,"pm25":6,"temperature":24,"update":{"installed_version":16777233,"latest_version":16777233,"state":"idle"},"update_available":null,"voc_index":344}'`

shrx, 2023-07-24:
```
debug 2023-07-24 23:05:12: Loaded state from file /home/pi/src/zigbee2mqtt-1.30.4/data/state.json
info  2023-07-24 23:05:12: Logging to console and directory: '/home/pi/src/zigbee2mqtt-1.30.4/data/log/2023-07-24.23-05-12' filename: log.txt
debug 2023-07-24 23:05:12: Removing old log directory '/home/pi/src/zigbee2mqtt-1.30.4/data/log/2023-02-28.19-20-15'
info  2023-07-24 23:05:12: Starting Zigbee2MQTT version 1.30.4-dev (commit #307e0dca)
info  2023-07-24 23:05:12: Starting zigbee-herdsman (0.14.115)
debug 2023-07-24 23:05:12: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/home/pi/src/zigbee2mqtt-1.30.4/data/coordinator_backup.json","databaseBackupPath":"/home/pi/src/zigbee2mqtt-1.30.4/data/database.db.backup","databasePath":"/home/pi/src/zigbee2mqtt-1.30.4/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
info  2023-07-24 23:05:14: zigbee-herdsman started (resumed)
info  2023-07-24 23:05:14: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20211115,"transportrev":2},"type":"zStack12"}'
debug 2023-07-24 23:05:14: Zigbee network parameters: {"channel":11,"extendedPanID":"0x00124b001ccc696c","panID":6754}
info  2023-07-24 23:05:14: Currently 8 devices are joined:
info  2023-07-24 23:05:14: zarnica (0xb4e3f9fffe6c1d64): LED1732G11 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, white spectrum, opal white (Router)
info  2023-07-24 23:05:14: daljinc (0x0c4314fffe4f287d): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
info  2023-07-24 23:05:14: vremeNoter (0x00124b0025047b63): SNZB-02 - SONOFF Temperature and humidity sensor (EndDevice)
info  2023-07-24 23:05:14: vremeZunaj (0x00124b00250478ff): SNZB-02 - SONOFF Temperature and humidity sensor (EndDevice)
info  2023-07-24 23:05:14: steker (0x84b4dbfffe96d245): E1603/E1702/E1708 - IKEA TRADFRI control outlet (Router)
info  2023-07-24 23:05:14: vremeKopalnica (0x00124b0025048713): SNZB-02 - SONOFF Temperature and humidity sensor (EndDevice)
info  2023-07-24 23:05:14: zarnicaL1 (0xec1bbdfffe5463b1): ZL1000100-CCT-US-V1A02 - Linkind Zigbee LED 9W A19 bulb, dimmable & tunable (Router)
info  2023-07-24 23:05:14: windstrike (0x3425b4fffea8592c): E2112 - IKEA Vindstyrka air quality and humidity sensor (Router)
info  2023-07-24 23:05:14: Zigbee: disabling joining new devices.
info  2023-07-24 23:05:14: Connecting to MQTT server at mqtt://localhost
debug 2023-07-24 23:05:14: Using MQTT anonymous login
info  2023-07-24 23:05:15: Connected to MQTT server
info  2023-07-24 23:05:15: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
debug 2023-07-24 23:05:15: Received MQTT message on 'zigbee2mqtt/bridge/info' with data '{"commit":"307e0dca","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":11,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"ikea_ota_use_test_url":false,"last_seen":"disable","legacy_api":true,"legacy_availability_payload":true,"log_directory":"/home/pi/src/zigbee2mqtt-1.30.4/data/log/%TIMESTAMP%","log_file":"log.txt","log_level":"info","log_output":["console","file"],"log_rotation":true,"log_symlink_current":false,"log_syslog":{},"output":"json","pan_id":6754,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"availability":{},"blocklist":[],"device_options":{},"devices":{"0x00124b00250478ff":{"friendly_name":"vremeZunaj"},"0x00124b0025047b63":{"friendly_name":"vremeNoter"},"0x00124b0025048713":{"friendly_name":"vremeKopalnica"},"0x0c4314fffe4f287d":{"friendly_name":"daljinc","legacy":false},"0x3425b4fffea8592c":{"friendly_name":"windstrike","humidity_precision":1,"temperature_precision":1},"0x84b4dbfffe96d245":{"friendly_name":"steker"},"0xb4e3f9fffe6c1d64":{"availability":{"tumeout":1},"description":"TRADFRI E27 1000 lm, dimmable, white spectrum, opal white","friendly_name":"zarnica"},"0xec1bbdfffe5463b1":{"description":"Linkind A60 1","friendly_name":"zarnicaL1"}},"external_converters":[],"frontend":{"host":"0.0.0.0","port":8022},"groups":{"1":{"devices":["0xb4e3f9fffe6c1d64/1"],"friendly_name":"test"},"65289":{"devices":["0xb4e3f9fffe6c1d64/1","0xec1bbdfffe5463b1/1"],"friendly_name":"tradfri_scenes"}},"homeassistant":false,"map_options":{"graphviz":{"colors":{"fill":{"coordinator":"#e04e5d","enddevice":"#fff8ce","router":"#4ea3e0"},"font":{"coordinator":"#ffffff","enddevice":"#000000","router":"#ffffff"},"line":{"active":"#009900","inactive":"#994444"}}}},"mqtt":{"base_topic":"zigbee2mqtt","force_disable_retain":false,"include_device_information":false,"server":"mqtt://localhost"},"ota":{"disable_automatic_update_check":false,"ikea_ota_use_test_url":false,"update_check_interval":1440},"passlist":[],"permit_join":false,"serial":{"disable_led":false,"port":"/dev/ttyACM0"}},"config_schema":{"definitions":{"device":{"properties":{"debounce":{"description":"Debounces messages of this device","title":"Debounce","type":"number"},"debounce_ignore":{"description":"Protects unique payload values of specified payload properties from overriding within debounce time","examples":["action"],"items":{"type":"string"},"title":"Ignore  [... cut ...]
debug 2023-07-24 23:05:42: Saving state to file /home/pi/src/zigbee2mqtt-1.30.4/data/state.json
info  2023-07-24 23:05:42: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
info  2023-07-24 23:05:42: Disconnecting from MQTT server
info  2023-07-24 23:05:42: Stopping zigbee-herdsman...
info  2023-07-24 23:05:42: Stopped zigbee-herdsman
info  2023-07-24 23:05:42: Stopped Zigbee2MQTT
```

IKEA E2112 Vindstyrka device is publishing on the topic "zigbee2mqtt/windstrike".
both humidity and temperature resolutions are set to 1 in configuration.

Koenkk, 2023-07-25:
As can be seen in the log, the precision of the zigbee messages is 0:

```
debug 2023-07-24 23:05:19: Received Zigbee message from 'windstrike', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2700}' from endpoint 1 with groupID 0
-> this is 27.00 degrees
debug 2023-07-24 23:05:20: Received Zigbee message from 'windstrike', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":5700}' from endpoint 1 with groupID 0
-> this is 57% humidity
```

Zigbee2MQTT cannot increase the precision, only reduce it.

shrx, 2023-07-25:
It's weird that we can't get a better resolution for humidity and temperature readouts over zigbee, since the raw data with higher resolution is actually available over serial: https://github.com/just-oblivious/vindstyrka-docs

sjorge, 2023-08-21:
The SEN54 should support 0.1 steps, at least when talking to it over i2c, but doesn't look like the fw from ikea exposes that 😑

The display also can't display that so I doubt we will get it in a future update. 😞

rfenouil, 2023-09-01:
Is it a fantasy to imagine Ikea people will endup reading this thread and decide to modify one line of code in their firmware ?
If you work for Ikea, please make a sign here :smile: 
We all bought this device because of Z2M compatibility... Thanks ! :smiley:

sjorge, 2023-09-01:
There is/was one account u/TRADFRI on reddit that was run by someone from the team there. But they dont take suggestions.

201505409, 2023-09-02:
Well in that case. I created a pull request for the z2m documentation. So now it states it as the first thing under notes. Hopefully, IKEA will see this. At least this will inform potential buyers...

shrx, 2023-09-02:
Thanks for updating the documentation, I guess that's all we can do. Here is the above mentioned pull request https://github.com/Koenkk/zigbee2mqtt.io/pull/2202 

@Koenkk I guess we can close this?

closed 2023-09-03 by Koenkk