Koenkk/zigbee2mqtt #17637: [Wrong device]:

cloudbr34k84, 2023-05-10:
### Link

https://a.aliexpress.com/_m0uVjs8

### Model

PJ-MGW1203

### Description

Zigbee Single Phase Power Metering Module with an 80A CT Clamp

### Vendor

Tuya

### Picture (link)

https://smarthomescene.com/wp-content/uploads/2023/03/tuya-zigbee-clamp-energy-meter-pj-mgw1203-buy-smarthomescene.jpg.webp

### Database entry

Router,  0x70ac08fffe916c19, 0x40B3

### Notes

The energy sensor us not correct.  It's not a continue increase,  rather it delta value.

Koenkk, 2023-05-11:
Please post the `data/database.db` entry of this device.

cloudbr34k84, 2023-05-11:
Same as other comments.

cloudbr34k84, 2023-05-13:
Here is the Device about page 
[image]

cloudbr34k84, 2023-05-22:
@Koenkk updated with database.db info

Koenkk, 2023-05-23:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

tenkaibr, 2023-05-24:
I'm having a problem with this model. It already appears correctly but the current, power and energy data are null. linkquality and voltage are showing ok.

I'm using the latest version of edge (I just installed it) and the latest version of the coordinator.

```
debug 2023-05-24 01:35:48Received Zigbee message from '0x70ac08fffe3a46de', type 'commandMcuGatewayConnectionStatus', cluster 'manuSpecificTuya', data '{"payloadSize":194}' from endpoint 1 with groupID 0
debug 2023-05-24 01:35:53Received Zigbee message from '0x70ac08fffe3a46de', type 'commandMcuGatewayConnectionStatus', cluster 'manuSpecificTuya', data '{"payloadSize":195}' from endpoint 1 with groupID 0
debug 2023-05-24 01:35:58Received Zigbee message from '0x70ac08fffe3a46de', type 'commandMcuGatewayConnectionStatus', cluster 'manuSpecificTuya', data '{"payloadSize":196}' from endpoint 1 with groupID 0
debug 2023-05-24 01:36:04Received Zigbee message from '0x70ac08fffe3a46de', type 'commandMcuGatewayConnectionStatus', cluster 'manuSpecificTuya', data '{"payloadSize":197}' from endpoint 1 with groupID 0
debug 2023-05-24 01:36:09Received Zigbee message from '0x70ac08fffe3a46de', type 'commandMcuGatewayConnectionStatus', cluster 'manuSpecificTuya', data '{"payloadSize":198}' from endpoint 1 with groupID 0
debug 2023-05-24 01:36:14Received Zigbee message from '0x70ac08fffe3a46de', type 'commandMcuGatewayConnectionStatus', cluster 'manuSpecificTuya', data '{"payloadSize":199}' from endpoint 1 with groupID 0
debug 2023-05-24 01:36:20Received Zigbee message from '0x70ac08fffe3a46de', type 'commandMcuGatewayConnectionStatus', cluster 'manuSpecificTuya', data '{"payloadSize":200}' from endpoint 1 with groupID 0
debug 2023-05-24 01:36:25Received Zigbee message from '0x70ac08fffe3a46de', type 'commandMcuGatewayConnectionStatus', cluster 'manuSpecificTuya', data '{"payloadSize":201}' from endpoint 1 with groupID 0
debug 2023-05-24 01:36:27Received Zigbee message from '0x70ac08fffe3a46de', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":70}' from endpoint 1 with groupID 0
debug 2023-05-24 01:36:30Received Zigbee message from '0x70ac08fffe3a46de', type 'commandMcuGatewayConnectionStatus', cluster 'manuSpecificTuya', data '{"payloadSize":202}' from endpoint 1 with groupID 0
debug 2023-05-24 01:36:36Received Zigbee message from '0x70ac08fffe3a46de', type 'commandMcuGatewayConnectionStatus', cluster 'manuSpecificTuya', data '{"payloadSize":203}' from endpoint 1 with groupID 0
debug 2023-05-24 01:36:36Received Zigbee message from '0x70ac08fffe3a46de', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,4,254],"type":"Buffer"},"datatype":2,"dp":20}],"seq":52224}' from endpoint 1 with groupID 0
info 2023-05-24 01:36:36MQTT publish: topic 'zigbee2mqtt/0x70ac08fffe3a46de', payload '{"current":null,"energy":null,"linkquality":105,"power":null,"voltage":127.8}'
```

[image]

[image]

cloudbr34k84, 2023-05-24:
Have you tried re adding it??

tenkaibr, 2023-05-24:
> Have you tried re adding it??

Yeap. I've already lost count.

Is your device working properly?

cloudbr34k84, 2023-05-24:
> > Have you tried re adding it??
> 
> Yeap. I've already lost count.
> 
> Is your device working properly?

Yeah man 
[image]

tenkaibr, 2023-05-25:
> Yeah man !

That's weird. When did you buy your device? Mine just arrived. Maybe they changed something? Here it is blinking a red light every 2-3s. It's normal?

cloudbr34k84, 2023-05-25:
> > Yeah man !
> 
> That's weird. When did you buy your device? Mine just arrived. Maybe they changed something? Here it is blinking a red light every 2-3s. It's normal?

I bought months ago. Do you have a tuya zigbee hub to connect to to. Is it picking up any sensors?

tenkaibr, 2023-05-26:
> I bought months ago. Do you have a tuya zigbee hub to connect to to. Is it picking up any sensors?

Unfortunately I don't have a tuya zigbee hub but I already bought it to do the tests. The problem is that it won't arrive in time to open a claim.

Your log (debug) shows something like this:

`type 'commandMcuGatewayConnectionStatus', cluster 'manuSpecificTuya', data '{"payloadSize":203}' from endpoint 1 with groupID 0`

Should I open another topic to address the problem? This one is listed as closed.

And thanks for the attention!

cloudbr34k84, 2023-05-26:
Yeah you should open a separate issue. You need to attach herdsman logs.  Do you have that enabled?

tenkaibr, 2023-05-27:
I have no idea but I'll look into it. But first I will buy a tuya hub to confirm if the problem is in ziggbe2mqtt.

danwooller, 2025-01-08:
> > I'm having a problem with this model. It already appears correctly but the current, power and energy data are null. linkquality and voltage are showing ok.
> > I'm using the latest version of edge (I just installed it) and the latest version of the coordinator.
> > ```
> > debug 2023-05-24 01:35:48Received Zigbee message from '0x70ac08fffe3a46de', type 'commandMcuGatewayConnectionStatus', cluster 'manuSpecificTuya', data '{"payloadSize":194}' from endpoint 1 with groupID 0
> > debug 2023-05-24 01:35:53Received Zigbee message from '0x70ac08fffe3a46de', type 'commandMcuGatewayConnectionStatus', cluster 'manuSpecificTuya', data '{"payloadSize":195}' from endpoint 1 with groupID 0
> > debug 2023-05-24 01:35:58Received Zigbee message from '0x70ac08fffe3a46de', type 'commandMcuGatewayConnectionStatus', cluster 'manuSpecificTuya', data '{"payloadSize":196}' from endpoint 1 with groupID 0
> > debug 2023-05-24 01:36:04Received Zigbee message from '0x70ac08fffe3a46de', type 'commandMcuGatewayConnectionStatus', cluster 'manuSpecificTuya', data '{"payloadSize":197}' from endpoint 1 with groupID 0
> > debug 2023-05-24 01:36:09Received Zigbee message from '0x70ac08fffe3a46de', type 'commandMcuGatewayConnectionStatus', cluster 'manuSpecificTuya', data '{"payloadSize":198}' from endpoint 1 with groupID 0
> > debug 2023-05-24 01:36:14Received Zigbee message from '0x70ac08fffe3a46de', type 'commandMcuGatewayConnectionStatus', cluster 'manuSpecificTuya', data '{"payloadSize":199}' from endpoint 1 with groupID 0
> > debug 2023-05-24 01:36:20Received Zigbee message from '0x70ac08fffe3a46de', type 'commandMcuGatewayConnectionStatus', cluster 'manuSpecificTuya', data '{"payloadSize":200}' from endpoint 1 with groupID 0
> > debug 2023-05-24 01:36:25Received Zigbee message from '0x70ac08fffe3a46de', type 'commandMcuGatewayConnectionStatus', cluster 'manuSpecificTuya', data '{"payloadSize":201}' from endpoint 1 with groupID 0
> > debug 2023-05-24 01:36:27Received Zigbee message from '0x70ac08fffe3a46de', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":70}' from endpoint 1 with groupID 0
> > debug 2023-05-24 01:36:30Received Zigbee message from '0x70ac08fffe3a46de', type 'commandMcuGatewayConnectionStatus', cluster 'manuSpecificTuya', data '{"payloadSize":202}' from endpoint 1 with groupID 0
> > debug 2023-05-24 01:36:36Received Zigbee message from '0x70ac08fffe3a46de', type 'commandMcuGatewayConnectionStatus', cluster 'manuSpecificTuya', data '{"payloadSize":203}' from endpoint 1 with groupID 0
> > debug 2023-05-24 01:36:36Received Zigbee message from '0x70ac08fffe3a46de', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,4,254],"type":"Buffer"},"datatype":2,"dp":20}],"seq":52224}' from endpoint 1 with groupID 0
> > info 2023-05-24 01:36:36MQTT publish: topic 'zigbee2mqtt/0x70ac08fffe3a46de', payload '{"current":null,"energy":null,"linkquality":105,"power":null,"voltage":127.8}'
> > debug 2023-05-24 03:00:08Received Zigbee message from '0x70ac08fffe3a46de', type 'attributeReport', cluster 'genBasic', data '{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000f\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0012"}' from endpoint 1 with groupID 0
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > ![image](https://private-user-images.githubusercontent.com/39348260/240487800-3bd18969-6cf1-4cc2-8400-baa20452e7c0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYyNzkwMTMsIm5iZiI6MTczNjI3ODcxMywicGF0aCI6Ii8zOTM0ODI2MC8yNDA0ODc4MDAtM2JkMTg5NjktNmNmMS00Y2MyLTg0MDAtYmFhMjA0NTJlN2MwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA3VDE5MzgzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWViM2M4NjhhM2UxZmViNDlmNDhmOGQ5ZWUxMGI3ZjIwNDhhNjc0MjA3YTE5MDAzZDExZmEyNzc3NjNiZWViMTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ql3VJXnkqmNSGc8NV6ZS6XbCpxdwds9fJrgH3TSIK-E)
> > ![image](https://private-user-images.githubusercontent.com/39348260/240487842-bff7d715-c2f5-4245-b0d4-e0c47d2dc881.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYyNzkwMTMsIm5iZiI6MTczNjI3ODcxMywicGF0aCI6Ii8zOTM0ODI2MC8yNDA0ODc4NDItYmZmN2Q3MTUtYzJmNS00MjQ1LWIwZDQtZTBjNDdkMmRjODgxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA3VDE5MzgzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk2ZWYwYWVkZWY5ZDI4MWY5NWViMjBiY2M4MzYzNDE0ZTM2ZDdhYTYxOTViNjZiNTk3YjliZWY1MmQzMzE5MDAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.2dVvvKrA5VS5qcwzQEkkztGK2uJmBgkLze92EqyTAmc)
> 
> I have the same thing here, did you solve it already?

I just have 0 on energy.