Koenkk/zigbee2mqtt #17638: [Wrong device]:

cloudbr34k84, 2023-05-10:
### Link

https://www.smarthome.com.au/product/aeotec-smart-plug/

### Model

GP-WOU019BBEWA

### Description

Samsung SmartThings AU Smart Plug

### Vendor

Samsung Smartthings

### Picture (link)

https://www.auditech.com.au/content/images/thumbs/0005343_samsung-smartthings-smart-plug-gp-wou019bbewa-white.jpeg

### Database entry

Router,  0x680ae2fffe7d3708, 0xE8C4

### Notes

The devices does not expose current and voltage sensors

Koenkk, 2023-05-11:
Please post the data/database.db entry of this device.

cloudbr34k84, 2023-05-11:
> Please post the data/database.db entry of this device.

Hey,  I would if I had them.  My database file does not have any of my devices. 
[image]

Koenkk, 2023-05-12:
You are looking at the `database.db.backup`, it should be `database.db`

cloudbr34k84, 2023-05-12:
> You are looking at the `database.db.backup`, it should be `database.db`

I don't see that file man??

Koenkk, 2023-05-13:
In that case, provide me the `Zigbee Manufacturer` from the z2m frontend -> device -> about page.

cloudbr34k84, 2023-05-13:
> In that case, provide me the `Zigbee Manufacturer` from the z2m frontend -> device -> about page.

[image]

Koenkk, 2023-05-13:
This plug doesn't support current and voltage measurements: 

- https://github.com/Koenkk/zigbee-herdsman-converters/blob/ababa9d9d7a20a1af75309d7778b0d534e05121f/devices/smartthings.js#L134
- https://github.com/Koenkk/zigbee2mqtt/issues/5198

I will update the device page with this info.

closed 2023-05-13 by Koenkk

cloudbr34k84, 2023-05-13:
> This plug doesn't support current and voltage measurements:
> 
> * https://github.com/Koenkk/zigbee-herdsman-converters/blob/ababa9d9d7a20a1af75309d7778b0d534e05121f/devices/smartthings.js#L134
> * [Samsung SmartThings Plug (GP-WOU019BBDWG) is not reporting voltage or current #5198](https://github.com/Koenkk/zigbee2mqtt/issues/5198)
> 
> I will update the device page with this info.

That's the UK plug,  no the Australian plug.

Koenkk, 2023-05-14:
@cloudbr34k84 but the firmware/hardware in it is probably the same.

cloudbr34k84, 2023-05-14:
> @cloudbr34k84 but the firmware/hardware in it is probably the same.

No that's fine,  im just making sure the information will be updated.