Koenkk/zigbee2mqtt #17639: Wrong Information

cloudbr34k84, 2023-05-10:
### Link

https://a.aliexpress.com/_mOanxNe

### Model

SM-O301-AZ

### Description

AU 4 Plug 10A Power Board + USB

### Vendor

Tuya

### Picture (link)

https://www.google.com/imgres?imgurl=https%3A%2F%2Fae01.alicdn.com%2Fkf%2FH1ded5755178c4b2b8028a9848ed38924f%2FUseeLink-Smart-Power-Strip-Au-Zigbee-Power-Bar-Multiple-Outlet-Extension-Cord-with-4USB-and-4.jpg_640x640.jpg&tbnid=IgbBVtwL8N7OhM&vet=1&imgrefurl=https%3A%2F%2Fwww.aliexpress.com%2Fi%2F4000504463608.html&docid=9D0Pn7QoR8sJQM&w=640&h=640&hl=en-AU&source=sh%2Fx%2Fim

### Database entry

Router,  0x70b3d52b60016c72, 0x3361

Koenkk, 2023-05-11:
Please post the `data/database.db` entry of this device

cloudbr34k84, 2023-05-11:
Same comment as before,  my database file does not have the entries?

cloudbr34k84, 2023-05-13:
Here is the device about page
[image]

Koenkk, 2023-05-14:
@nikosillo  you added `_TZ3000_o005nuxx` in https://github.com/Koenkk/zigbee-herdsman-converters/pull/1882, can you confirm your `_TZ3000_o005nuxx` carries model number `SM-SO306EZ-10`?

cloudbr34k84, 2023-05-18:
I have updated the information with my Database.db file info.

Koenkk, 2023-05-20:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-05-20 by Koenkk

cloudbr34k84, 2023-05-25:
@Koenkk the image doesn't match what i supplied
[image]

Koenkk, 2023-05-25:
Added the picture!