Koenkk/zigbee2mqtt #17641: [New device support]: Develco/Frient Smart Plug Mini 2 (Type F) SPLZB-141

westlund, 2023-05-11:
### Link

https://www.develcoproducts.com/products/smart-plugs/smart-plug-mini/

### Database entry

{"id":54,"type":"Router","ieeeAddr":"0x0015bc002f0142c8","nwkAddr":16544,"manufId":4117,"manufName":"frient A/S","powerSource":"Mains (single phase)","modelId":"SPLZB-141","epList":[2,1,242],"endpoints":{"1":{"profId":49353,"epId":1,"devId":1,"inClusterList":[5,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":81,"inClusterList":[0,2,3,4,5,6,9,1794,2820,2821],"outClusterList":[3,10,25,1030],"clusters":{"genBasic":{"attributes":{"modelId":"SPLZB-141","manufacturerName":"frient A/S","powerSource":1,"zclVersion":7,"dateCode":"2023-02-15 14:43"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"dateCode":"2023-02-15 14:43","zclVersion":7,"interviewCompleted":true,"meta":{},"lastSeen":1683787630034,"defaultSendRequestWhen":"immediate"}

### Comments

I started implementing according to the new device instructions with the -131 definition as template but ran into problems. As this is the first time for me, takeoff distance is probably longer than if someone with previous experience takes on the task. I will be happy to assist!

emandtf, 2023-05-13:
It's not very difficult to test a new variant especially in this case that seems 99% similar to 131.

You just need to edit (DO A BACKUP OF THE ORIGINAL ONE) "/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/develco.js" file.
For a very-quick-but-dirty test you can just exchange this rows:
`zigbeeModel: ['SPLZB-131'],
model: 'SPLZB-131',`
with:
`fingerprint: [{modelID: 'SPLZB-131', manufacturerName: 'SPLZB-131'}, {modelID: 'SPLZB-141', manufacturerName: 'SPLZB-141'}],
model: 'SPLZB-131/141',`

then:

1. save file
2. do a zigbee2mqtt restart
3. it could be needed to remove and then add again your new 141 from zigbee2mqtt web-interface

westlund, 2023-05-14:
Thanks, I actually started with the develco.js file as an external converter, the file containing everything from Develco with an added entry for the SPLZB-141. It is a copy of the existing SPLZB-131 entre with 131 replaced with 141.

The initial declarations in the file (like `const exposes = require('../lib/exposes');` ) seem to refer to the wrong path so it was a no go when reloading z2m, see log output below.

I'm running zigbe2mqtt as an add-on in hassio but need guidance on which is the correct path to work with or refer to. 

```
[12:04:11] INFO: Starting Zigbee2MQTT...
node:internal/modules/cjs/loader:1029
  throw err;
  ^
Error: Cannot find module '../lib/exposes'
Require stack:
- /app/dist/util/utils.js
- /app/dist/util/settings.js
- /app/index.js
    at Function.Module._resolveFilename (node:internal/modules/cjs/loader:1026:15)
    at Function.Module._load (node:internal/modules/cjs/loader:871:27)
    at Module.require (node:internal/modules/cjs/loader:1098:19)
    at require (node:internal/modules/cjs/helpers:108:18)
    at /app/data/extension/externally-loaded.js:1:17
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:150:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:157:12)

```

emandtf, 2023-05-14:
> I actually started with the develco.js file as an external converter

I never created those "external converters" so I cannot help you so much.
I've suggested the most rapid (but I know it's dirty) way to test your 141 against 131 features.

It seems you duplicated Develco.js file in another path and then added a new section for your 141........but dependencies are RELATIVE to Develco.js file and path, so this is the issue you get.

Sorry but I can support you only for my (dirty) way.

westlund, 2023-05-14:
Thank you for helping me so far @emandtf . I'll try finding the right docker container or file path to be able to update the builtin develco.js file - but I'm still familiarising with the hassio architecture.

westlund, 2023-05-14:
Ok, I got it working with the following definition added straight into the LowerDir of the docker image at 

`/mnt/data/docker/overlay2/f6b1f42108e70389bbfbc346833169ed56c9a3bd63ad6961169375c91e3f4841/diff/app/node_modules/zigbee-herdsman-converters/devices/develco.js`

The definition added is below and the plug can be controlled and the properties seem to be accurate at a first inspection.

[image]

```
    {
        zigbeeModel: ['SPLZB-141'],
        model: 'SPLZB-141',
        vendor: 'Develco',
        description: 'Power plug',
        fromZigbee: [fz.on_off, develco.fz.electrical_measurement, develco.fz.metering],
        toZigbee: [tz.on_off],
        exposes: [e.switch(), e.power(), e.current(), e.voltage(), e.energy(), e.ac_frequency()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(2);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await reporting.onOff(endpoint);
            await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
            await reporting.activePower(endpoint);
            await reporting.rmsCurrent(endpoint);
            await reporting.rmsVoltage(endpoint);
            await reporting.readMeteringMultiplierDivisor(endpoint);
            await reporting.currentSummDelivered(endpoint);
            await reporting.acFrequency(endpoint);
        },
        endpoint: (device) => {
            return {default: 2};
        },
    }
```

Koenkk, 2023-05-15:
@westlund thanks, added the device.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-05-16 by westlund